OpenKinect/libfreenect #430: [Stream 70] Expected max 1748 data bytes, but got 1908. Dropping...

dfamil, 2015-01-06:
i ma getting this error with the sample SimpleViewer. I have not be able to get it to work with openFrameworks yet. 

[Stream 70] Expected max 1748 data bytes, but got 1908. Dropping...
[Stream 70] Expected max 1748 data bytes, but got 1908. Dropping...

when i get this error with processing onenNI1 and libfreenect 1 it does it fro a few lines and then works. is this  libfreenect or openni2 problem? is there a way to get it to keep trying?

piedar, 2015-01-07:
Looks similar to #29.  Can you give the full output of `freenect-glview`?  And maybe try with a more powerful computer.

dfamil, 2015-01-07:
hi
freenect-glview works fine it is the SimpleView is in the openni2 sample folder. I am using a 2010 mac pro so speed sold not  be a issue.

d

piedar, 2015-01-07:
On Linux, SimpleView gives that message only very rarely, and everything works despite it.  I will test again on OSX when I get a chance.

dfamil, 2015-01-08:
hi
there is another issue that i discovered starting with 10.9 the headers are no longer part of glut or opengl frameworks. They are in the sdks in the application: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk/System/Library/Frameworks/GLUT.framework

when do you think you will have the ir stream working with the kinect?

piedar, 2015-01-08:
Do the moved headers cause a build error?  I recently [updated](https://github.com/OpenKinect/libfreenect/commit/e551c9afc92721c00b4e537f05621f600fb44486#diff-2a223b2dbe4b8adba24dcfb102869d01) the examples to make CMake handle finding OpenGL and GLUT rather than just adding "-framework OpenGL - framework GLUT" to the linker flags.  Hopefully the CMake devs are keeping up with all these recent changes in OSX.

piedar, 2015-01-08:
Oh, and IR works for me.  Press 'f' twice in freenect-glview to see it.  And in code you would do

    freenect_set_video_mode(f_dev, freenect_find_video_mode(FREENECT_RESOLUTION_MEDIUM, FREENECT_VIDEO_IR_8BIT));

dfamil, 2015-01-08:
hi

yes i did get a error i had to enter the correct path after the error showed up in the buiild fil(forgot what it was called)

i meant the OpenNI2-FreenectDriver. I am using Openframeworks so from that theo states you can;t both have the ofxKinect installed and OpenNI. I am not sure if this applies to OpenNI2 which uses a different FreenectDriver than the base ofkKinect. 

d

dfamil, 2015-01-08:
hi ben

i guess i am confused i thought that glview uses the ibfreenect.dylib.

while openni2 uses ibfreenectDriver.dylib

this is what is wirtten in the ofcNI2  addon  

//	if (ir.setup(*device)) // only for xtion device (OpenNI2-FreenectDriver issue)
//	{
//		ir.setSize(320, 240);
//		ir.setFps(30);
//		ir.start();
//	}

piedar, 2015-01-08:
Ah yes, libfreenect itself supports IR, but OpenNI2-FreenectDriver does not have a way to hook it up.  I think I should add it in one of these days.