OpenKinect/libfreenect #433: Add ability to control IR projector intensity

piedar, 2015-01-12:
Add a function like

    freenect_set_ir_brightness(uint8_t brightness);

and implement it by sending a control command with parameter 0x0015.

Requested by @zhumxcq

zhumxcq, 2015-01-23:
Patch information

From 4e1603beb63d3117e22b2f3e8905da0ef5924213 Mon Sep 17 00:00:00 2001
From: Meng Xi Zhu <zhumxcq@gmail.com>
Date: Mon, 12 Jan 2015 20:17:04 -0500
Subject: [PATCH 1/2] Added function to change IR brightness

---
 include/libfreenect.h |  2 ++
 src/cameras.c         | 10 ++++++++++
 2 files changed, 12 insertions(+)

diff --git a/include/libfreenect.h b/include/libfreenect.h
index edc3898..b8e7e21 100644
--- a/include/libfreenect.h
+++ b/include/libfreenect.h
@@ -688,6 +688,8 @@ FREENECTAPI void freenect_set_fw_address_nui(freenect_context * ctx, unsigned ch
  */
 FREENECTAPI void freenect_set_fw_address_k4w(freenect_context * ctx, unsigned char * fw_ptr, unsigned int num_bytes);
 
+FREENECTAPI void freenect_change_ir_brightness(freenect_device *dev, uint16_t brightness);
+
 
 #ifdef __cplusplus
 }
diff --git a/src/cameras.c b/src/cameras.c
index 12eaba2..1c5dde1 100644
--- a/src/cameras.c
+++ b/src/cameras.c
@@ -1243,6 +1243,7 @@ int freenect_set_video_buffer(freenect_device *dev, void *buf)
 FN_INTERNAL int freenect_camera_init(freenect_device *dev)
 {
 	freenect_context *ctx = dev->parent;
+
 	int res;
 	res = freenect_fetch_reg_pad_info(dev);
 	if (res < 0) {
@@ -1285,3 +1286,12 @@ FN_INTERNAL int freenect_camera_teardown(freenect_device *dev)
 	freenect_destroy_registration(&(dev->registration));
 	return 0;
 }
+
+void freenect_change_ir_brightness(freenect_device *dev, uint16_t brightness){
+	if(dev){
+		int res = write_register(dev, 0x15, brightness);
+		if(res!=0){
+			//FN_ERROR("freenect_change_ir_brightness(): Failed to change IR brightness\n");
+		}
+	}
+}
-- 
1.9.1

From 806be4e0ee4f7174ea77943baf39caca52f80f9c Mon Sep 17 00:00:00 2001
From: Meng Xi Zhu <zhumxcq@gmail.com>
Date: Mon, 12 Jan 2015 20:22:03 -0500
Subject: [PATCH 2/2] added comment for header function

---
 include/libfreenect.h | 5 +++++
 1 file changed, 5 insertions(+)

diff --git a/include/libfreenect.h b/include/libfreenect.h
index 0e93a4b..f504ffc 100644
--- a/include/libfreenect.h
+++ b/include/libfreenect.h
@@ -689,6 +689,11 @@ FREENECTAPI void freenect_set_fw_address_nui(freenect_context * ctx, unsigned ch
  */
 FREENECTAPI void freenect_set_fw_address_k4w(freenect_context * ctx, unsigned char * fw_ptr, unsigned int num_bytes);
 
+/**
+ * Allows the user to specify the brightness of the IR image
+ *
+ * @param brightness value between 0 (lowest brightness) - 50 (highest brightness)
+ */
 FREENECTAPI void freenect_change_ir_brightness(freenect_device *dev, uint16_t brightness);
 
 
-- 
1.9.1

floe, 2015-02-11:
Could you please send this as a pull request on Github?

zhumxcq, 2015-02-11:
Not sure how to do that exactly...

floe, 2015-02-11:
@zhumxcq First, you have to create your own fork of libfreenect on github. For the rest, there's a good tutorial available here: https://help.github.com/articles/creating-a-pull-request/

piedar, 2015-02-11:
I was going to merge it manually, but I have been lazy...

piedar, 2015-02-11:
Brightness must be in the range 1 - 50.

piedar, 2015-10-16:
Needs [documentation update](https://github.com/OpenKinect/libfreenect/commit/b6c2eb33845ccd98e6496dd172048f659ebc57a7#commitcomment-13095562).

piedar, 2015-10-18:
Updated documentation in ad5b98d.

commit b6c2eb33 (2015-02-11): Add ability to control IR projector brightness - fixes #433
commit 68073c37 (2015-02-13): Return int from freenect_get_ir_brightness() - fixes #433
closed 2015-10-18 by piedar