OpenKinect/libfreenect #434: Unable to compile on Ubuntu 14.04

davivcgarcia, 2015-02-03:
I'm trying to compile the code tag 0.5.2 on Ubuntu 14.04.1 LTS but CMake fails to resolve the dependencies. Checking logs, it is complaining about the -pthreads flag used, but this seems to be not available on my platform. CMake is able to find without problems the lib required for -pthread but the compilation fails.

LuisLlana, 2015-10-14:
I have the same problem with Debian 9

piedar, 2015-10-15:
It could be a [compiler warning](https://stackoverflow.com/a/25130590/1925996) caused by [bug 15058](https://cmake.org/Bug/view.php?id=15058) in CMake < 3.1.

Try this workaround:

    cmake -DCMAKE_C_FLAGS="-Wno-old-style-definition" ..

If that fails, please post the build log.

LuisLlana, 2015-10-15:
I have been able to compile it. The problem is that there were unfulfilled dependencies.

piedar, 2017-10-10:
Assumed fixed by newer CMake.

closed 2017-10-10 by piedar