OpenKinect/libfreenect #435: Cannot build with audio support

locatw, 2015-02-15:
I'm trying to build libfreenect with audio support on Ubuntu 14.04 (x86), but an error occurred when extracting SystemUpdate.zip in make step.

    $ cmake -DBUILD_PYTHON=ON -DBUILD_CV=ON -DPYTHON_EXECUTABLE=/usr/bin/python2 -DBUILD_REDIST_PACKAGE=OFF -L ..
    $ make
    Scanning dependencies of target firmware
    Downloading SystemUpdate.zip from http://www.xbox.com/system-update-usb
    Reading response...
    done, saved to SystemUpdate.zip
    Extracting $systemupdate/FFFE07DF00000001 from system update file...
    Traceback (most recent call last):
      File "/home/loca/build/libfreenect/src/fwfetcher.py", line 548, in <module>
        pirs = extractPirsFromZip(fw)
      File "/home/loca/build/libfreenect/src/fwfetcher.py", line 538, in extractPirsFromZip
        pirs = z.open("$systemupdate/FFFE07DF00000001").read()
      File "/usr/lib/python2.7/zipfile.py", line 961, in open
        zinfo = self.getinfo(name)
      File "/usr/lib/python2.7/zipfile.py", line 909, in getinfo
        'There is no item named %r in the archive' % name)
    KeyError: "There is no item named '$systemupdate/FFFE07DF00000001' in the archive"
    make[2]: *** [src/CMakeFiles/firmware] error 1
    make[1]: *** [src/CMakeFiles/firmware.dir/all] error 2
    make: *** [all] error 2

Downloading SystemUpdate.zip was successful and file was saved at build/src/ directory, and a file "FFFE07DF00000001" exists in unzipped directory.

Why this error occurs?

My python environment is (extracts from cmake output log):

    -- Found PythonInterp: python2 (found version "2.7.6") 
    -- Found PythonLibs: /usr/lib/i386-linux-gnu/libpython2.7.so (found version "2.7.6") 

Thank you

fabian247, 2015-02-26:
I ran into the same problem. It's due to a misspelling in fwfetcher.py or because Microsoft renamed a path in their SystemUpdate.zip.

fwfetcher.py refers to `$systemupdate/FFFE07DF00000001` in line 538. The MS file SystemUpdate.zip extracts to `$Systemupdate`. Note the capital 'S'.

You can correct the foldername in fwfetcher.py or you can also repack SystemUpdate.zip and rename the folder inside the archive.

Hope this helps.

locatw, 2015-02-28:
I can build successfully when I correct the line 538 in fwfetcher.py. Thanks!

piedar, 2015-03-01:
Reopened to remind me to fix it.  Thanks for finding the solution, @fabian247

commit 95ae9707 (2015-03-02): Fix capitalization in fwfetcher.py - fixes #435