OpenKinect/libfreenect #437: Build successful, can't connect to kinect

daviddanyi, 2015-02-25:
Greetings!
I built the libfreenect driver for a virtual windows 7 x64 (and later also on the host machine, which runs windows 8.1 x64). The build went without errors and the sample codes also compiled. I also installed the drivers for the xbox NUI devices (motor, camera, audio).
The problem comes when i run the sample applications (for example glview). It detects the kinect no problem, i receive the 1 device found message, but after that it exits with " Could not claim interface on camera: -3".
I looked up the -3 error code of libisb, and it means ACCESS DENIED. Itried running the program as administrator and giving all priviliges to the user account, but it made no difference. The exact same thing happened on both the virtual machine and the host computer.

Do any of you have an idea what should I do next?

piedar, 2015-02-25:
The Windows documentation is out of date; I have been meaning to update it.

I found success by using [Zadig](http://zadig.akeo.ie/) to install the libusbK driver for each subdevice.  Then I patched and compiled libusb as described in [this comment](https://github.com/OpenKinect/libfreenect/issues/405#issuecomment-67560293).

daviddanyi, 2015-02-25:
Thanks, that did the trick. Sort of.
The glview test app started up alright, the IR projector started working and the tilt motor levelled the kinect, though I didn't get any image back. It says in the terminal window that the isochron communication failed.

piedar, 2015-02-25:
-12 is LIBUSB_ERROR_UNSUPPORTED, so maybe cmake is still pointing to the unpatched libusb?

daviddanyi, 2015-03-04:
Correct again. I did a clean make and build and it started working.
The only problem is: it only works for a few seconds and after that the stream is lost.
I read in the aforementioned thread that setting the log level to low fixes the problem. It certeanly improved the situation, but still the max tima was 10-15 seconds.
Could this be caused by the fact that i'm using a virtual machine?

I really appreciate your help, and am sorry for not responding for a week.

piedar, 2015-03-10:
It could very well be due to the VM, especially if it has poor USB support.

piedar, 2017-07-22:
There's now a proposed libusb patch for [isochronous transfers via WinUSB](https://github.com/libusb/libusb/pull/284).  I've heard it's better than libusbk for Windows 8.1 and above.

closed 2017-07-22 by piedar