OpenKinect/libfreenect #439: No device found

Znafon, 2015-04-01:
Hye. I'm trying to use freenect but i does not find the device.

I installed the libfreenect package available in fedora 21 repositories (version 0.5.0) and connect the device. While it's successfully detected by the OS and in `/dev/bus/usb/001/006` it's not found by `freenect-glview` which display `Number of devices found: 0`.

floe, 2015-04-02:
Most of the time, this is caused by some permissions issue. Have you tried running freenect-glview as root?

Znafon, 2015-04-02:
Yes. I've digged further in the issue and the kinect I have is not seen as one. 
It's product number is not registered at this line of `usb_libusb10.c`: `if (desc.idVendor == VID_MICROSOFT && (desc.idProduct == PID_NUI_CAMERA || desc.idProduct == PID_K4W_CAMERA))`

Znafon, 2015-04-04:
I tried to change the ID in the source code. Here is what i got when i plugged the kinect:

    ➜  libfreenect git:(master) ✗ lsusb | grep 045e
    Bus 002 Device 003: ID 045e:02c4 Microsoft Corp. 
    Bus 002 Device 002: ID 045e:02d9 Microsoft Corp. 
    Bus 001 Device 002: ID 045e:02d9 Microsoft Corp. 

It tried to change `#define PID_K4W_CAMERA 0x02bf` to both `#define PID_K4W_CAMERA 0x02d9` and `#define PID_K4W_CAMERA 0x02c4` but `freenect-glview` gives now this error message:

    Kinect camera test
    Number of devices found: 2
    Could not open camera: -3
    Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.Could not open device

floe, 2015-04-06:
Ah, now I get it: this is the Kinect One, right? In this case, you will need https://github.com/OpenKinect/libfreenect2 which is a separate project.

closed 2015-04-09 by piedar