OpenKinect/libfreenect #447: Raspberry Pi 2: Failed to open motor/audio subdevice or it is not disabled.

vladimiroltean, 2015-07-10:
Hi,

I'm having issues trying to run some examples (**freenect-glview, freenect-chunkview, freenect-cppview, freenect-hiview**), with all of them returning:
> Kinect camera test
> Number of devices found: 1
> Could not open audio: -1
> Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.Could not open device

Occasionally, libusb seems to return error code *-4* too. These errors are not permissions-related.
Note that **freenect-regview** does function correctly.
What actually bothers me is that, for an unknown reason, I am unable to run the [OpenCV example](http://openkinect.org/wiki/C%2B%2BOpenCvExample) on the Raspberry Pi 2, it returns the same error codes and messages (*-4, could not open motor and audio*).
The thing is, I do not actually care for motor and audio, only video and depth.
I have been able to run the OpenCV example on an Ubuntu 14.04 laptop, but I have no idea why it does not run on the Raspberry Pi 2 (I have already followed these steps and also asked for help [here](http://www.mariolukas.de/2015/04/proof-of-concept-3d-scanner-with-kinect-and-raspberry-pi2/)). I have been running the latest version of both Raspbian and Arch Linux, with no success other than **freenect-regview**.
A strange thing that happens is, whenever I run a freenect application, the following message appears in **dmesg**, and I don't know whether it is a feature or a bug:
> [ 9513.700007] usb 2-1.1.1: reset high-speed USB device number 76 using ehci-pci
> [ 9518.790646] usb 2-1.1.1: can't restore configuration #1 (error=-110)
> [ 9518.791096] usb 2-1.1.1: USB disconnect, device number 76
> [ 9518.878488] usb 2-1.1.1: new high-speed USB device number 77 using ehci-pci
> [ 9524.428664] usb 2-1.1.1: new high-speed USB device number 78 using ehci-pci
> [ 9524.523389] usb 2-1.1.1: New USB device found, idVendor=045e, idProduct=02be
> [ 9524.523399] usb 2-1.1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ 9524.523406] usb 2-1.1.1: Product: Microsoft Kinect Audio, \xffffffc2\xffffffa9\xffffffa9 2011 Microsoft Corporation. All rights reserved.
> [ 9524.523412] usb 2-1.1.1: Manufacturer: Microsoft
> [ 9524.523416] usb 2-1.1.1: SerialNumber: A22592W00346213A
I (think I) have a Kinect for Windows device, which has been adapted to USB via a Microsoft 1431 12V 1.08A power adapter. However, plugging the device into a powered USB hub has not made any difference. For sake of completeness, here is the relevant part of the output of **lsusb**:
> Bus 002 Device 060: ID 045e:02bf Microsoft Corp.
> Bus 002 Device 078: ID 045e:02be Microsoft Corp.
> Bus 002 Device 054: ID 045e:02c2 Microsoft Corp.

pascalgross, 2015-07-14:
Any progress here? Got the same problem on an Xilinx Zynq running an embedded Linux (PetaLinux)

piedar, 2015-07-14:
It looks like audio is [required for motor control](https://github.com/OpenKinect/libfreenect/blob/89f77f6d2c23876936af65766a4c140898bc3ea8/src/usb_libusb10.c#L352) on the K4W.  If you only need the camera, you can modify [libfreenect.hpp#191](https://github.com/OpenKinect/libfreenect/blob/89f77f6d2c23876936af65766a4c140898bc3ea8/wrappers/cpp/libfreenect.hpp#L191) to

    freenect_select_subdevices(m_ctx, static_cast<freenect_device_flags>(FREENECT_DEVICE_CAMERA));

But I'm not sure why the error happens in the first place.  It's possible that some USB hosts can't support whatever is required for audio.  Maybe @wizgrav or @ofTheo knows something more?

vladimiroltean, 2015-07-14:
Thank you, this indeed solves the issue for me.

a155935498, 2015-07-20:
I modify libfreenect.hpp#191 to "freenect_select_subdevices(m_ctx, static_cast<freenect_device_flags>(FREENECT_DEVICE_CAMERA));" but it doesn't work out.
The problem(Could not open audio: -4) is still exist.
Is there anyone knows how to fix it?

FollowingT, 2015-07-28:
I also encountered the same problem. And i failed to fix it by the solution mentioned above. It still returned  the error "Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.Could not open device"
Is there anyone give me a favor?

piedar, 2015-08-04:
Anyone facing this problem: please try updating to dcb2929.  It should allow continuing execution even if audio device fails to load.

sebastiangonzalezbotasi, 2016-03-02:
Hello, I am Sebastian, I am from Argentina. To compile the example KinextExample You have but same error.

Any new solution?

piedar, 2016-08-25:
Hopefully fixed by #479 - please reopen if it still doesn't work.

closed 2016-08-25 by piedar