OpenKinect/libfreenect #449: udev rules and Raspberry Pi 2

matthiasplappert, 2015-07-25:
Hey everyone, I'm using libfreenect on my Raspberry Pi 2. Everything works great, except for one small inconvenience: I can only use libfreenect if I run my Python script as root (`sudo python script.py`).

I'm pretty sure that my issue is related to this one: https://github.com/OpenKinect/libfreenect/issues/445. When I checked, this is what I found:
```
ls /etc/udev/rules.d/
40-scratch.rules  99-com.rules
```
So the rules for freenect where indeed missing. I attempted to fix this by copying the udev rules (https://github.com/OpenKinect/libfreenect/blob/master/platform/linux/udev/51-kinect.rules). Here's what it looks like right now:
```
ls /etc/udev/rules.d/ -l
total 12
-rw-r--r-- 1 root root 114 May  6 23:20 40-scratch.rules
-rw-r--r-- 1 root root 599 Jul 25 16:11 51-kinect.rules
-rw-r--r-- 1 root root 338 May  6 22:39 99-com.rules
```
However, I'm still not able to use libfreenect as a non-root user. I tried to reboot and the likes, no luck. Any idea what I'm missing?

piedar, 2015-07-26:
Is your user a member of the `plugdev` group?  Could there be a conflicting udev rule?  It could also depend on how your distribution handles device permissions.

What are your OS and Kinect model?

matthiasplappert, 2015-07-27:
> Is your user a member of the `plugdev` group?
```
groups pi
pi : pi adm dialout cdrom sudo audio video plugdev games users netdev input spi i2c gpio
```
Hence yes, my user (`pi`) is a member of `plugdev`.

> Could there be a conflicting udev rule?
I'm not really sure how udev rules work. Here are all my rules:
```
cat /etc/udev/rules.d/40-scratch.rules
ATTRS{idVendor}=="0694", ATTRS{idProduct}=="0003", SUBSYSTEMS=="usb", ACTION=="add", MODE="0666", GROUP="plugdev"

cat /etc/udev/rules.d/51-kinect.rules
# ATTR{product}=="Xbox NUI Motor"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02b0", MODE="0666"
# ATTR{product}=="Xbox NUI Audio"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02ad", MODE="0666"
# ATTR{product}=="Xbox NUI Camera"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02ae", MODE="0666"

# Kinect for Windows
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02c2", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02be", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02bf", MODE="0666"

cat /etc/udev/rules.d/99-com.rules
SUBSYSTEM=="gpio*", PROGRAM="/bin/sh -c 'chown -R root:gpio /sys/class/gpio && chmod -R 770 /sys/class/gpio; chown -R root:gpio /sys/devices/virtual/gpio && chmod -R 770 /sys/devices/virtual/gpio'"
SUBSYSTEM=="input", GROUP="input", MODE="0660"
SUBSYSTEM=="i2c-dev", GROUP="i2c", MODE="0660"
SUBSYSTEM=="spidev", GROUP="spi", MODE="0660"
```
I tried to remove 99-com.rules, but that does not change things either.

> What are your OS and Kinect model?
Raspberry Pi 2 B with latest RASPBIAN and linux kernel 4.0.9. Kinect model 1414.

Here's the output from `lsusb` as well:
```
lsusb
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp.
Bus 001 Device 004: ID 7392:7811 Edimax Technology Co., Ltd EW-7811Un 802.11n Wireless Adapter [Realtek RTL8188CUS]
Bus 001 Device 005: ID 0409:005a NEC Corp. HighSpeed Hub
Bus 001 Device 007: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 001 Device 006: ID 045e:02b0 Microsoft Corp. Xbox NUI Motor
Bus 001 Device 008: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
```

Thanks for your help! Any more ideas how to resolve this?

piedar, 2015-07-27:
I'm stumped - what you have there ought to work.  Maybe there's something useful in the [udev log](https://unix.stackexchange.com/questions/36251/in-which-log-should-i-check-for-udev-errors).

matthiasplappert, 2015-07-27:
This is the only log related to udev that I could find (after setting the log level to `debug`, reloading all rules and running my python script that triggers the permission error):
```
Jul 27 17:28:23 raspberrypi udevd[177]: udevd message (SET_LOG_PRIORITY) received, log_priority=7
Jul 27 17:28:33 raspberrypi udevd[177]: udevd message (RELOAD_RULES) received
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/10-local-rpi.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/etc/udev/rules.d/40-scratch.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/42-qemu-usb.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/50-udev-default.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/etc/udev/rules.d/51-kinect.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/55-dm.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-cdrom_id.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-fuse.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-gnupg.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-i2c-tools.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-ifplugd.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-libgphoto2-2.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-alsa.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-input.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-serial.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-storage-dm.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-storage-tape.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-storage.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-persistent-v4l.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-python-pifacecommon.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-python3-pifacecommon.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/60-triggerhappy.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/61-accelerometer.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/run/udev/rules.d/61-dev-root-link.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/64-xorg-xkb.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/70-udev-acl.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/75-cd-aliases-generator.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/75-net-description.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/75-persistent-net-generator.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/75-probe_mtd.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/75-tty-description.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/78-sound-card.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/80-drivers.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/80-networking.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/80-udisks.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/85-hwclock.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/85-usbmuxd.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/90-alsa-restore.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/91-permissions.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/95-keyboard-force-release.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/95-keymap.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/lib/udev/rules.d/95-udev-late.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: reading '/etc/udev/rules.d/99-com.rules' as rules file
Jul 27 17:28:33 raspberrypi udevd[177]: rules use 141696 bytes tokens (11808 * 12 bytes), 23055 bytes buffer
Jul 27 17:28:33 raspberrypi udevd[177]: temporary index used 38520 bytes (1926 * 20 bytes)
```
Seems to me that everything is fine.

If we can't figure it out it's not that much of a problem, since running my script as sudo is perfectly fine in my environment.

hntuan94, 2016-01-14:
Hi matthiasplappert!
How many frame per second (depth) can Rasperry Pi get from kinect?

matthiasplappert, 2016-01-14:
I think I got 30fps on an Raspberry Pi 2.

MaxConners, 2018-02-10:
I had this same problem, but then I came accross [this post](http://cyberalaska.cs.uaf.edu/blogs/archives/268) from 2012.  

The temporary fix is to `sudo chmod 777 /dev/bus/usb/001/*`

A permanent fix is to `sudo nano /etc/udev/rules.d/66-kinect.rules`

paste this:

` # ATTR{product}=="Xbox NUI Motor"
 SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02b0", MODE="0666"
 # ATTR{product}=="Xbox NUI Audio"
 SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02ad", MODE="0666"
 # ATTR{product}=="Xbox NUI Camera"
 SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02ae", MODE="0666"`

finally, restart udev: `sudo /etc/init.d/udev restart`

Hope it works for you ;)

Cheers