OpenKinect/libfreenect #451: Tilt & LED - Ubuntu 14.04 - Kinect 1473

senia907, 2015-08-09:
Hi,
I'm using Ubuntu 14.04 with Kinect(Xbox 360) model 1473.
I compiled and installed the latest libfreenect.
running "freenect-glview" shows the video streams, yet with a strange error:
"Could not open audio: -4", And I can't control the Tilt motor and LED.
lsusb shows:
Bus 003 Device 007: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 003 Device 023: ID 045e:02ad Microsoft Corp. Xbox NUI Audio
Bus 003 Device 004: ID 045e:02c2 Microsoft Corp. 
(No NUI Motor as in previous models).
This is not a new problem(https://github.com/OpenKinect/libfreenect/issues/376), 
yet I searched everywhere and couldn't fix it.
(I need to grab .rawlog videos using MRPT, yet its not working, and I think its connected.)
Anyone knows how to fix it?
Thanks.

piedar, 2015-08-09:
I believe audio firmware is required to control the motor of that model.  The easiest way I know is to add the [ofxKinectExtras](https://github.com/openframeworks/openFrameworks/tree/master/addons/ofxKinect/src/extra) to your project and [load it into libfreenect](https://github.com/OpenKinect/libfreenect/issues/376#issuecomment-41211251).

senia907, 2015-08-09:
Thanks for the fast comment, yet as I mentioned, I don't use freenect as a project, is there any other way or I'm in trouble just because I have the 1473 model?

piedar, 2015-08-09:
I think then you are stuck, unless you can modify `freenect-glview` to load firmware and compile it as C++.  Or figure out some other way to load the firmware.  (Or find a model 1414.)  Until recently it would fail completely in this case, so I am glad to hear that the video shows up at least.

senia907, 2016-02-18:
Sorry for bumping, left it for a while. How can I load firmware to the glview and what do you mean by "firmware"? And how does the [ofxKinect ](https://github.com/openframeworks/openFrameworks/tree/master/addons/ofxKinect/src/extra) could help me? I see only 2 functions in there(getFWSize1473(); and getFWSizek4w();) which I couldn't find any reference to them (or similar) in freenect and particularly in the "glview" program.
It's very important for me as auto exposure doesn't work neither.

piedar, 2016-02-18:
The firmware is executable code that runs on the Kinect.  Microsoft owns that code and we don't have an open-source alternative.  libfreenect provides a way to send the firmware to the Kinect, but it would be legally risky to provide the firmware itself.

The fwfetcher.py program is one workaround; it downloads and extracts the firmware from a freely-available XBox360 system update package.  But it only works for model 1414.

For the newer models, the solution is to compile the firmware into a byte array and upload it at runtime.  You can do so by modifying your program to pass a pointer to the byte array to libfreenect, like [these instructions](https://github.com/OpenKinect/libfreenect/issues/376#issuecomment-41211251) demonstrate.  You would need to add that code into glview and also make sure the ofxKinectExtras.cpp file is added to the compilation.  It will also be necessary to compile the program as C++ since the ofxKinectExtras.h file is C++.

I wish we could just ship the firmware, but the legal situation is too murky for me.