OpenKinect/libfreenect #452: Kinect 1473 on linux Mint 17 - demos shutting down

libasoles, 2015-08-30:
Hi. 
I've buyed the Kinect 1473. And I understand it's not as easy to setup like 1414. 

I'm not being able to run my kinect more than a few seconds.

I'm using Linux Mint 17 64bits on a AMD x2 2.9GHZ PC. 

I'm connecting the kinect to PC via USB, and also to the wall (220v).

Green light is always blinking.

If I write 'lsusb' sometimes I get these two items but sometimes just one (changes all the time):
Bus 001 Device 093: ID 045e:02ae Microsoft Corp. Xbox NUI Camera
Bus 001 Device 091: ID 045e:02ad Microsoft Corp. Xbox NUI Audio

I've compiled libfreenect, added rules to proper folder, added myself to proper groups, blacklisted gspca_kinect, and tried demos.

Most than all, generated errors where "no audio device". Also sometimes kinect doesn't run at all:
Error: Invalid index [0]
Error: Kinect not connected?

I also use to get "invalid magic" and "lost packages" errors.

And finally, the "something went terrible wrong" or just:
USB camera marked dead, stopping streams
send_cmd: Output control transfer failed (-4)

As far as I understand, there's a problem with USB 2 bootleneck so I run:
sudo sh -c "echo -1 > /sys/module/usbcore/parameters/autosuspend"

Also, I searched the repo for solutions to a couple of problems with 1473. And I tried applying this pull request:
https://github.com/OpenKinect/libfreenect/pull/325

And also I edited "src/usb_libusb10.c" and removed this line (ref: https://github.com/OpenKinect/libfreenect/commit/f6cb19e3f0607dcb77fd413db64887b17cf2f956):
memset(strm, 0, sizeof(*strm));

Well, I'm still having problems running the demos, and the demos crashes after a few seconds. Only solved issue is the non audio found issue. 

Let me show you some logs:

./build/bin/cppview 
Failed to submit isochronous transfer 0: -1

./build/bin/glpclview
USB device disappeared, cancelling stream 82 :(
USB camera marked dead, stopping streams
send_cmd: Output control transfer failed (-4)

./build/bin/glview 
Kinect camera test
Number of devices found: 0

./build/bin/glview (once again)
... (works for a while)
[Stream 70] Invalid magic 83ff
[Stream 70] Lost 6 packets
[Stream 70] Lost too many packets, resyncing...
...
[Stream 70] Inconsistent flag 75 with 239 packets in buf (242 total), resyncing...
'w'-tilt up, 's'-level, 'x'-tilt down, '0'-'6'-select LED mode, 'f'-video format
'e' - auto exposure, 'b' - white balance, 'r' - raw color, 'm' - mirror
 raw acceleration:    0    0    0  mks acceleration: 0.000000 0.000000 0.000000iso_callback(): failed to resubmit transfer after successful completion: -4
USB camera marked dead, stopping streams
write_register: 0x0005 <= 0x00
send_cmd: Output control transfer failed (-4)
write_register: 0x0006 <= 0x00
send_cmd: Output control transfer failed (-4)

shutting down streams...
-- done!

./build/bin/hiview 
freenect_fetch_reg_info: send_cmd read 4 bytes (expected 118)
[Stream 70] Lost too many packets, resyncing...
[Stream 70] Invalid magic ffff
...
iso_callback(): failed to resubmit transfer after successful completion: -4
USB camera marked dead, stopping streams
write_register: 0x0005 <= 0x00

send_cmd: Output control transfer failed (-4)
Something went terribly wrong.  Aborting.

I found some pull requests and forks about audio.bin, and SystemUpdate.zip, and fixes, most of them by @ofTheo (eg: https://github.com/ofTheo/Kinect1473/blob/master/README.md)
But I'm not sure how to apply them and also I understand main issues are with OS X, not mentioning linux. 

So, I'm kinda lost. Here are some questions, just to know:

1) Should green light keep steady in order to make kinect work?

2) Will 'lsusb' ever show the kinect "motor" as device? 

I'm not sure what to expect when trying things. 

Well, that's it. Any clue?

Thanks.

ofTheo, 2015-08-30:
This might be helpful for you: https://github.com/openframeworks/openFrameworks/tree/master/addons/ofxKinect/src/extra

There is a function in libfreenect to load the firmware via a memory address. The files above have the firmware as a function.

libasoles, 2015-08-30:
I'm looking for that function. I have no experience with c but maybe I can manage. No results by now anyway.

I'd like to understand what the issue is. The firmware that is downloaded by libfreenect (from http://www.xbox.com/system-update-usb) is not the correct one for 1473? And is the only one provided by Microsoft (latest one)?

piedar, 2015-08-31:
Here are the [instructions](https://github.com/OpenKinect/libfreenect/issues/376#issuecomment-41211251) for loading the firmware into memory.  You are correct that the downloaded firmware is for the 1414 model and does not work with 1473.

libasoles, 2015-08-31:
Thanks both.
I'm not yet in context because I'm not a c/c++ developer. Actually I'm planning to make some sketches using Processing. So would you mind putting me in orbit please? 

When ofTheo says call my functions "after freenect_init", he's referring to a OpenFrameworks app, right? And specifically to ofxKinect addon? Then I should create a openframeworks project just to flash the correct firmware (only once). Is this right?

Thanks for your efforts.

libasoles, 2015-09-01:
I was saying nonsense, wasn't I?

ofTheo code should go in cpp wrapper, right? After this line:
if(freenect_init(&m_ctx, NULL) < 0)

Now I see. And kinectContext is actually m_ctx.

Well, it's not working for me yet, but I'll keep on trying.

libasoles, 2015-09-01:
I must say I have some achievements. Most of the errors are gone.

Also, after applying the above code, I read somewhere that I should run the micview example first in order to flash the audio firmware. And indeed something happened: 

Trying to open ./audios.bin as firmware...
Found firmware image
...
Firmware successfully uploaded and launched.  Device will disconnect and reenumerate.

Demo runs for a while and then sadly it stops.

piedar, 2015-09-01:
Try deleting audios.bin, since that is the wrong firmware for your device.

libasoles, 2015-09-02:
Ok, I've tried deleting audios.bin. But it's the same:

audio: received an iso IN packet of strange length: 164
USB device disappeared, canceling stream 82 :(
USB audio marked dead, stopping streams

Now I'm trying to compile audios.bin by myself, as this user did: https://github.com/OpenKinect/libfreenect/pull/362#issuecomment-36455468

So, digging into low level programming :S

ofTheo, 2015-09-02:
all you should need to do is call: 

        freenect_set_fw_address_nui(kinectContext, ofxKinectExtras::getFWData1473(), ofxKinectExtras::getFWSize1473());
        freenect_set_fw_address_k4w(kinectContext, ofxKinectExtras::getFWDatak4w(), ofxKinectExtras::getFWSizek4w());

immediately after freenect_init  

you would need to include ofxKinectExtras.h and .cpp and ( which I posted a link above ) and pass in the context to both functions as the first argument.

libasoles, 2015-09-03:
Sure, but where? I tried that, adding the lines in the examples (in examples/micview.c & glview.c, and of course including the cpp file). And I get compilation errors (sometimes "unknown type name ‘class’" , other times something about <iostream> in the header file).

Also, I need to know if this is a "one time process". I mean, flashing the audios firmware. Because if it is not, I don't see the way to use this code with Processing. 

Thanks for being there. Sorry if I'm not seeing something obvious. But believe me I'm trying.

alwynmathew, 2017-10-02:
@ofTheo in which file exactly should we call this function?

> and call:
> 
> ```
> freenect_set_fw_address_nui(kinectContext, ofxKinectExtras::getFWData1473(), ofxKinectExtras::getFWSize1473());
> freenect_set_fw_address_k4w(kinectContext, ofxKinectExtras::getFWDatak4w(), ofxKinectExtras::getFWSizek4w());
> ```
> 
> after `freenect_init`

aaalgo, 2018-12-23:
I'm having exactly the same issue.  No clue at all what happened.
Then I changed a computer and the error disappeared and everything was good.
Both computers are running ubuntu 18.04 and libfreenect installed from apt-get.

The failing one is an HP desktop and the working on is a lenovo x1 carbon laptop.