OpenKinect/libfreenect #453: Meaning of depth numbers

a155935498, 2015-09-05:
I try to run "demo_cv_sync.py" 
and get a depth array from "  freenect.sync_get_depth() ".

But, I have no idea about the meaning of  numbers in this array.
I  want to know the unit of  depth numbers.
Can someone teach me please.
Or,  where can i find the information.

piedar, 2015-09-07:
According to [this conversation](https://groups.google.com/forum/#!msg/openkinect/WkG5_sQsImY/RzqSs64aveEJ), FREENECT_DEPTH_MM will give you "the 
distance from the Kinect's optical centre to the object in millimeters".  They also mention how it is possible to convert from the raw depth values to distance manually, for greater precision.

closed 2016-05-19 by piedar

ChinhLeVan, 2019-03-25:
i can't be found