OpenKinect/libfreenect #454: Mac OS X: USB Device only partially registered

florianletsch, 2015-09-21:
I have installed the freenect drivers using homebrew and have had success with this before. However, when I now connect my device, it isn't recognized by any of the demo commands (i.e. `freenect-regview`).

I use `ioreg -p IOUSB` to list any USB devices on my system. 

The following shows up when the Kinect is connected to USB:

```
  | |   +-o Xbox NUI Motor@fa132000  <class IOUSBDevice, id 0x10000055a, registered, matched, active, busy 0 (147 ms), retain 9>
```

However, what **doesn't** show up (but what I am expecting from previous successful tries) would be the following:

```
  | | +-o HubDevice@fa130000  <class IOUSBHubDevice, id 0x100003cd2, registered, matched, active, busy 0 (9 ms), retain 15>
  | |   +-o Xbox NUI Motor@fa132000  <class IOUSBDevice, id 0x100003cdc, registered, matched, active, busy 0 (221 ms), retain 9>
  | |   +-o Xbox Kinect Audio, © 2011 Microsoft Corporation. All rights reserved.@fa131000  <class IOUSBDevice, id 0x100003ce9, registered, matched, active, busy 0 (7 ms), retain $
  | |   +-o Xbox NUI Camera@fa133000  <class IOUSBDevice, id 0x100003cf1, registered, matched, active, busy 0 (31 ms), retain 9>
```

It seems like only part of the Kinect is successfully registered as a USB device. How can this be? Any ideas how I can troubleshoot this?

RaitoBezarius, 2016-01-03:
I have the same problem on Arch Linux on the latest version of libfreenect.

armandok, 2018-05-21:
I am having the same problem on macOS 10.13.4 (17E202).