OpenKinect/libfreenect #461: Kinects scanning through water

Siggurd, 2016-01-12:
Hey, I am currently working on my Master Thesis in Geoinformatics. The goal is to scan objects submerged under water using both versions of Microsoft's Kinect sensor.

I conducted several experiments and found that the water surface is recognized as kind of a solid border once the Kinect is approached closer than ~50cm to the water and there are no points returned at all (both Kinect versions).
I also mounted the Kinects in an aquarium and submerged it a few cm underneath the water surface --> also no results!

Did you encounter this problem when you worked with the Kinect or do you have any idea why I am having these issues? Is there probably a fix that could make the Kinect see water not as solid border when it is close?

It would be great if you could help me here!

floe, 2016-01-13:
Simple reason: water absorbs IR light, so the signal-noise ratio will get a lot worse under water (see https://en.wikipedia.org/wiki/Electromagnetic_absorption_by_water)

Apart from that, you may also experience strong reflection from the water surface that overloads the sensor. Both are pretty fundamental physical issues...

Siggurd, 2016-01-13:
I am aware of these limitations. But still it is possible to get decent bathymetric data when the sensor is situated about 50+cm from the water surface (I tested this!), which of course needs to be corrected due to refraction. So apparently the sensor is powerful enough to penetrate water to a certain degree, but not when it is closer than its "normal" minimal range...

floe, 2016-01-14:
I really think it's the reflection. Did you have the sensor oriented perpendicular to the water surface, or at an angle?

Siggurd, 2016-01-14:
I tried it either way...no changes unfortunately!

closed 2017-07-22 by piedar