OpenKinect/libfreenect #462: Could not open audio: -4

kowalski100, 2016-02-04:
After building the source with the mentioned procedure in Readme.md, i am trying to get RGB and dept image from connected kinect sensors using the following python code.
`#import the necessary modules
import freenect
import cv2
import numpy as np
 
#function to get RGB image from kinect
def get_video():
    array,_ = freenect.sync_get_video()
    array = cv2.cvtColor(array,cv2.COLOR_RGB2BGR)
    return array
 
#function to get depth image from kinect
def get_depth():
    array,_ = freenect.sync_get_depth()
    array = array.astype(np.uint8)
    return array
 
if __name__ == "__main__":
    while 1:
        #get a frame from RGB camera
        frame = get_video()
        #get a frame from depth sensor
        depth = get_depth()
        #display RGB image
        cv2.imshow('RGB image',frame)
        #display depth image
        cv2.imshow('Depth image',depth)
 
        # quit program when 'esc' key is pressed
        k = cv2.waitKey(5) & 0xFF
        if k == 27:
            break
    cv2.destroyAllWindows()`

Executing the code gives the following error:
Could not open audio: -4
after this nothing happens except the cursor blinks.

piedar, 2016-02-05:
Sounds like firmware is missing.  If you're only interested in video/depth, change [this line](https://github.com/OpenKinect/libfreenect/blob/df5ab32879a348abf30f0f959606a1958afc7ff7/wrappers/python/freenect.pyx#L309) to

    freenect_select_subdevices(ctx, <freenect_device_flags> (FREENECT_DEVICE_CAMERA))

Then rebuild the python binding.

    cd wrappers/python
    cython freenect.pyx
    python2 setup.py build_ext --inplace

kowalski100, 2016-02-05:
Thanks for your kind response sir...
Yes that certainly solved the problem.. But the other thing that troubles me is that the program remains  stuck at `freenect.sync_get_video()`. looks like its waiting for kinect to respond... Nothing happens afterword... The drivers are all connected properly in VMwere.

piedar, 2016-02-06:
Your code works perfectly for me with Kinect model 1414.  Can you try it directly instead of through VMware?

kowalski100, 2016-02-09:
Well Now I tried it on Raspberry pi 2 Model B..
Well problem remains same, infect now `freenect-glview` gives me the following error..
```
Kinect camera test
Number of devices found: 1
freenect-glview: symbol lookup error: /usr/local/lib/libfreenect.so.0.5: undefined symbol: libusb_get_parent
```
I had the same error at `make` command that i solved using the following issue link..
http://stackoverflow.com/questions/28835794/undefined-reference-to-libusb-get-parent-compiling-freenect

Now everything successfully compiles but `freenect-glview` command gives the same libusb_get_parent error..

piedar, 2016-02-09:
Looks like it compiled against the new libusb version but is still running against the old version.  Try reloading the library cache with

    sudo ldconfig

You could also try

    LD_LIBRARY_PATH="${LD_LIBRARY_PATH}:/path/to/new/libusb/" freenect-glview