OpenKinect/libfreenect #464: Errors During Python Install

duffyevan, 2016-02-14:
Hello, every time I try to install the modules to python I get this error:
/usr/bin/ld: cannot find -lfreenect
/usr/bin/ld: cannot find -lfreenect_sync

Im trying to install to a raspberry pi B+ with jessie. All other steps have gone fairly well until trying to use and of the libraries. I get trouble using c++ as well...
Any info would be great.

piedar, 2016-02-16:
I'm guessing you followed these steps:

    cd wrappers/python
    python2 setup.py build_ext --inplace

That assumes that you've already got the core library installed.  Instead, try building python with the rest of the project:

    cd build
    cmake .. -DBUILD_PYTHON=ON
    make

Now you should find the python libs in `build/wrappers/python/`  Try the test program:

    python2 wrappers/python/python2/test.py

duffyevan, 2016-02-16:
Ahhh ok that makes sense, however when I went to run cmake it gives me this error:

CMake Error at wrappers/python/CMakeLists.txt:44 (message):
    Unable to find consistent Python 2 libraries.  Python interpreter is 2.7.9, but libraries are 3.4.2.

Any input? I can try to figure it out myself later but I'm busy now.
Thanks in advance

piedar, 2016-02-16:
OK, maybe you don't have Python 2 libraries installed.  Try

    cmake .. -DBUILD_PYTHON=OFF -DBUILD_PYTHON3=ON

duffyevan, 2016-02-16:
Well now it says python3 interpreter not found. I ran apt-get install python python3 and it says they're both the newest version... Looking at the first error it seems like I might just have the wrong libraries installed, Maybe I can try to install the python 2.7.9 libraries, would that work? This is a puzzler...

piedar, 2016-02-16:
Hmm, maybe you need CMake version at least 3.0 or python-dev and/or python3-dev packages.  I can only guess based on this [AskUbuntu question](https://askubuntu.com/questions/479260/cmake-can-not-find-pythonlibs) because I don't use Debian.

duffyevan, 2016-02-16:
Well, something magical happened, I tried to install those packages, they both said they were up to date, and then I tried cmake again and it worked fine and built the python stuff. Now the problem is, when I run "python2 setup.py build_ext  --inplace" I get the error:

/usr/bin/ld: cannot find -lfreenect
/usr/bin/ld: cannot find -lfreenect_sync
collect2: error: ld returned 1 exit status
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

It seems like its looking for the files before they're installed... Not sure, what do you think?

piedar, 2016-02-16:
You do not need those steps if CMake worked.  Just run `make` in the build directory and the python libraries will be in `wrappers/python/python2/` and `wrappers/python/python3/`.

duffyevan, 2016-02-16:
So i figured out what magically happened. I must have accidentally set the build_python constants to off so it built and installed a bunch of things to /usr/local/bin include and lib, but it didn't build the python stuff. So I set the constants back to on and building python didn't work again. I know C++ so I tried making a C++ program that just includes the library and prints "hi" and that worked so I might just try it with C++. I would really like to get python working thought... 
The issue with the python stuff is I don't know why it doesn't like the 2 different versions of python. I'll try uninstalling python 3 to try to remove the conflict

hbgxsm95, 2017-03-01:
Hi!  I met the same problem as you .  Would you please tell me whether you solve it? if not , could you please tell me how?

jonaseltes, 2017-03-08:
Also faced with the same problem. Any progress on this?

hbgxsm95, 2017-03-08:
Hey I solved it by "sudo apt-get install python-dev"   You may have a try.

zmistrz, 2018-03-28:
Hey,
Sorry for my english.

( hbgxsm95 ) when you install from repo:
" Hey I solved it by "sudo apt-get install python-dev" You may have a try." - is good solution.

when you install from source:
1. After use make and make install
2. Go to libfreenect/wrappers/python and execute
   python2 setup.py build_ext --inplace
3. Then you must copy shared object to your python version
sudo cp libfreenect/wrappers/python/freenect.so /usr/lib/python2.7/
or your project
cp libfreenect/wrappers/python/freenect.* ~/your_project/

I don't now how install  (.so) drivers in python, make and make install and ldconfig dosen't work ;-(

hbgxsm95, 2018-04-22:
Sorry It has been a lot time and I forgot the details.  Good luck!