OpenKinect/libfreenect #467: Device not found

ragasi92, 2016-05-05:
I install freenect in my raspberry pi 3 and when I try to test the kinect with freenect-glview show:
Kinect camera test
Number of devices found: 0

More info:

pi@raspberrypi:~ $ sudo dmesg -C
pi@raspberrypi:~ $ dmesg
[  144.508990] usb 1-1.2: new high-speed USB device number 8 using dwc_otg
[  144.609255] usb 1-1.2: New USB device found, idVendor=0409, idProduct=005a
[  144.609277] usb 1-1.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  144.611280] hub 1-1.2:1.0: USB hub found
[  144.611742] hub 1-1.2:1.0: 3 ports detected
[  145.198999] usb 1-1.2.2: new full-speed USB device number 9 using dwc_otg
[  145.337032] usb 1-1.2.2: New USB device found, idVendor=045e, idProduct=02b0
[  145.337052] usb 1-1.2.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  145.337065] usb 1-1.2.2: Product: Xbox NUI Motor
[  145.337078] usb 1-1.2.2: Manufacturer: Microsoft

I hope you can help me. Thank you

parkerlreed, 2016-05-19:
Check to see if your user is in the plugdev group

````
groups
````

If not add your user

````
usermod -a -G plugdev username
````

pierrep, 2017-03-14:
I'm having the exact same problem on an Xbox 360 1414 kinect. 

I've tried the "echo -1 | sudo tee -a /sys/module/usbcore/parameters/autosuspend" but it makes no difference. The device doesn't show up on lsusb. 

Also the vendors Id's showing up don't match up with the regular Udev rules...

e.g  idVendor=0409, idProduct=005a

Anyone know what's going on?

piedar, 2017-03-15:
@pierrep 

First start watching the kernel log.

    dmesg -wH

Then plug it in and see if any messages appear.  If nothing happens, check the power cable and double-check udev permissions.  Also try `sudo freenect-camtest`; if that works it's almost definitely permissions.  If all else fails, try a different USB port.

thillRobot, 2020-07-03:
I am having the same issue with Ubuntu 16.04(Mate) and an Xbox 360 1414 kinect. Has this issue been resolved?

flutterbrony, 2021-09-29:
Same here with a 1414 kinect too

nbdy, 2022-05-27:
Same issue with the 1414 here.

dmesg output when replugging:
```
[15203.363504] usb 3-1: USB disconnect, device number 9
[15203.363511] usb 3-1.2: USB disconnect, device number 10
[15205.529486] usb 1-3: new high-speed USB device number 48 using xhci_hcd
[15205.679571] usb 1-3: New USB device found, idVendor=0409, idProduct=005a, bcdDevice= 1.00
[15205.679578] usb 1-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[15205.740193] hub 1-3:1.0: USB hub found
[15205.740435] hub 1-3:1.0: 3 ports detected
[15206.345925] usb 1-3.2: new full-speed USB device number 49 using xhci_hcd
[15206.485845] usb 1-3.2: New USB device found, idVendor=045e, idProduct=02b0, bcdDevice= 1.07
[15206.485852] usb 1-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[15206.485855] usb 1-3.2: Product: Xbox NUI Motor
[15206.485857] usb 1-3.2: Manufacturer: Microsoft
[15408.487646] usb 1-3: USB disconnect, device number 48
[15408.487653] usb 1-3.2: USB disconnect, device number 49
[15415.446174] usb 1-3: new high-speed USB device number 50 using xhci_hcd
[15415.594699] usb 1-3: New USB device found, idVendor=0409, idProduct=005a, bcdDevice= 1.00
[15415.594706] usb 1-3: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[15415.628123] hub 1-3:1.0: USB hub found
[15415.628431] hub 1-3:1.0: 3 ports detected
[15416.265716] usb 1-3.2: new full-speed USB device number 51 using xhci_hcd
[15416.410079] usb 1-3.2: New USB device found, idVendor=045e, idProduct=02b0, bcdDevice= 1.07
[15416.410087] usb 1-3.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[15416.410089] usb 1-3.2: Product: Xbox NUI Motor
[15416.410091] usb 1-3.2: Manufacturer: Microsoft
```

udev permissions:
```
# ATTR{product}=="Xbox NUI Motor"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02b0", MODE="0666"
# ATTR{product}=="Xbox NUI Audio"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02ad", MODE="0666"
# ATTR{product}=="Xbox NUI Camera"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02ae", MODE="0666"

# Kinect for Windows
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02c2", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02be", MODE="0666"
SUBSYSTEM=="usb", ATTR{idVendor}=="045e", ATTR{idProduct}=="02bf", MODE="0666"
```

groups:
```
nbdy adm sudo plugdev lpadmin libvirt docker
```

the 
```
echo -1 | sudo tee -a /sys/module/usbcore/parameters/autosuspend
```
trick didn't work for me either.

does anyone have some hints?