OpenKinect/libfreenect #469: build failing due to xbox.com blocking non-standard user agents

melvinvermeeren, 2016-05-16:
Libfreeneect build fails due Microsofts Xbox website blocking clients without
a valid user-agent.

Issue occurs in the file fwfetcher.py on line 534

The function getFileOrURL on line 522 fails when trying to retrieve the file
from the given URL:
https://github.com/OpenKinect/libfreenect/blob/master/src/fwfetcher.py#L522

This is a problem as it is trying to download the SystemUpdate.zip file from 
Xbox website :
https://github.com/OpenKinect/libfreenect/blob/master/src/fwfetcher.py#L566

The Request function is called with only a URL as a parameter:
https://github.com/OpenKinect/libfreenect/blob/master/src/fwfetcher.py#L534

Possible fix:
Adding user agent to the Request call 

req = Request(url, headers={'User-Agent': 'Mozilla/5.0'})

parkerlreed, 2016-05-18:
I tried modifying the request line as mentioned and it still bails out shortly after in the compile. Not sure if it actually pulled the firmware or not.

bmilde, 2016-05-19:
Modifying the line: 

req = Request(url) to 

req = Request(url, headers={'User-Agent': 'Mozilla/5.0'}) 

in fwfetcher.py did the trick for me!

(Mac OS X El Capitan)

parkerlreed, 2016-05-19:
Sweet after the latest commit and this it worked great!

commit 903d2599 (2016-05-19): Add user agent in fwfetcher.py
Fixes #469
closed 2016-05-19 by piedar