OpenKinect/libfreenect #470: Build failure on Arch x86_64

parkerlreed, 2016-05-18:
There was a recent update to GCC6. Not sure if that broken anything. 

````
-- Operating system is Linux
-- Got System Processor x86_64
-- Linux x86_64 Detected
-- libfreenect will be installed to /usr
-- Headers will be installed to /usr/include/libfreenect
-- Libraries will be installed to /usr/lib
-- Found libusb-1.0:
--  - Includes: /usr/include/libusb-1.0
--  - Libraries: /usr/lib64/libusb-1.0.so
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.11") 
-- Found PythonLibs: /usr/lib64/libpython2.7.so (found version "2.7.11") 
-- Found PythonInterp: /usr/bin/python3 (found version "3.5.1") 
-- Found PythonLibs: /usr/lib64/libpython3.so (found version "3.5.1") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/parker/build/libfreenect-git/src/libfreenect-git
[ 10%] Built target fakenect
[ 18%] Built target freenectstatic
[ 33%] Built target freenect
Scanning dependencies of target freenect-micview
Scanning dependencies of target FreenectDriver
Scanning dependencies of target freenect-glview
Scanning dependencies of target freenect-regview
[ 35%] Building C object examples/CMakeFiles/freenect-micview.dir/micview.c.o
[ 36%] Building C object examples/CMakeFiles/freenect-glview.dir/glview.c.o
[ 38%] Building CXX object OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/DepthStream.cpp.o
[ 40%] Building C object examples/CMakeFiles/freenect-regview.dir/regview.c.o
[ 41%] Linking C executable ../bin/freenect-micview
Downloading SystemUpdate.zip from http://www.xbox.com/system-update-usb
Failed to reach download server.  Reason: Forbidden
Reading response...
Traceback (most recent call last):
  File "/home/parker/build/libfreenect-git/src/libfreenect-git/src/fwfetcher.py", line 567, in <module>
    "http://www.xbox.com/system-update-usb")
  File "/home/parker/build/libfreenect-git/src/libfreenect-git/src/fwfetcher.py", line 544, in getFileOrURL
    retval = response.read()
UnboundLocalError: local variable 'response' referenced before assignment
src/CMakeFiles/firmware.dir/build.make:57: recipe for target 'src/CMakeFiles/firmware' failed
make[2]: *** [src/CMakeFiles/firmware] Error 1
CMakeFiles/Makefile2:121: recipe for target 'src/CMakeFiles/firmware.dir/all' failed
make[1]: *** [src/CMakeFiles/firmware.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 43%] Building CXX object OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/DeviceDriver.cpp.o
[ 43%] Built target freenect-micview
[ 45%] Building CXX object OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/ColorStream.cpp.o
[ 46%] Linking C executable ../bin/freenect-glview
[ 48%] Linking C executable ../bin/freenect-regview
[ 48%] Built target freenect-regview
[ 48%] Built target freenect-glview
In file included from /home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/DepthStream.cpp:2:0:
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/DepthStream.hpp:20:55: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::DepthStream::DIAGONAL_FOV’ of non-integral type [-fpermissive]
     static const float DIAGONAL_FOV = 70 * (M_PI / 180);
                                                       ^
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/DepthStream.hpp:21:59: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::DepthStream::HORIZONTAL_FOV’ of non-integral type [-fpermissive]
     static const float HORIZONTAL_FOV = 58.5 * (M_PI / 180);
                                                           ^
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/DepthStream.hpp:22:57: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::DepthStream::VERTICAL_FOV’ of non-integral type [-fpermissive]
     static const float VERTICAL_FOV = 45.6 * (M_PI / 180);
                                                         ^
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/DepthStream.hpp:31:53: error: ‘constexpr’ needed for in-class initialization of static data member ‘const double FreenectDriver::DepthStream::ZERO_PLANE_PIXEL_SIZE_VAL’ of non-integral type [-fpermissive]
     static const double ZERO_PLANE_PIXEL_SIZE_VAL = 0.10520000010728836;
                                                     ^~~~~~~~~~~~~~~~~~~
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/DepthStream.hpp:32:54: error: ‘constexpr’ needed for in-class initialization of static data member ‘const double FreenectDriver::DepthStream::EMITTER_DCMOS_DISTANCE_VAL’ of non-integral type [-fpermissive]
     static const double EMITTER_DCMOS_DISTANCE_VAL = 7.5;
                                                      ^~~
OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/build.make:62: recipe for target 'OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/DepthStream.cpp.o' failed
make[2]: *** [OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/DepthStream.cpp.o] Error 1
[... 21 lines cut ...]
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/ColorStream.hpp:17:57: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::ColorStream::HORIZONTAL_FOV’ of non-integral type [-fpermissive]
     static const float HORIZONTAL_FOV = 62 * (M_PI / 180);
                                                         ^
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/ColorStream.hpp:18:57: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::ColorStream::VERTICAL_FOV’ of non-integral type [-fpermissive]
     static const float VERTICAL_FOV = 48.6 * (M_PI / 180);
                                                         ^
In file included from /home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/ColorStream.cpp:2:0:
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/ColorStream.hpp:16:57: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::ColorStream::DIAGONAL_FOV’ of non-integral type [-fpermissive]
     static const float DIAGONAL_FOV = 73.9 * (M_PI / 180);
                                                         ^
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/ColorStream.hpp:17:57: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::ColorStream::HORIZONTAL_FOV’ of non-integral type [-fpermissive]
     static const float HORIZONTAL_FOV = 62 * (M_PI / 180);
                                                         ^
/home/parker/build/libfreenect-git/src/libfreenect-git/OpenNI2-FreenectDriver/src/ColorStream.hpp:18:57: error: ‘constexpr’ needed for in-class initialization of static data member ‘const float FreenectDriver::ColorStream::VERTICAL_FOV’ of non-integral type [-fpermissive]
     static const float VERTICAL_FOV = 48.6 * (M_PI / 180);
                                                         ^
OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/build.make:110: recipe for target 'OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/ColorStream.cpp.o' failed
make[2]: *** [OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/ColorStream.cpp.o] Error 1
OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/build.make:86: recipe for target 'OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/DeviceDriver.cpp.o' failed
make[2]: *** [OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/src/DeviceDriver.cpp.o] Error 1
CMakeFiles/Makefile2:1037: recipe for target 'OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/all' failed
make[1]: *** [OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
````

parkerlreed, 2016-05-18:
Just realized this is most likely a dup of #469

piedar, 2016-05-19:
Thanks!  I guess using `const` there was a GCC extension that they no longer support.

closed 2016-05-19 by piedar

parkerlreed, 2016-05-19:
Cheers! Compiled successfully.

commit 5c1eb71c (2016-05-19): Use constexpr where appropriate
Fixes #470