OpenKinect/libfreenect #474: Callback function not called (Python)

JohnnytheRocket, 2016-07-14:
I have earlier successfully used libfreenect's python wrapper with a Raspberry Pi 2 (Raspbian), and I decided to try and up the performance of my program with an Odroid C2 (Ubuntu 16) instead. However I run into an issue I haven't seen before (and didn't manage to google up).

Basically I can call freenect.runloop(...) without issues: no errors and the IR-laser starts running, but the only callback(?) function that's being called is the

`def body(*args)`
`    if not keep_running:`
`       raise freenect.Kill`

method, which loops indefinitely, and neither the depth method nor the rgb method are called even once.

I have tried both my own code that worked with the Raspberry, and the demo_cv_async example with the same result. Any ideas?

piedar, 2016-07-15:
Is there any console output?  Try running `freenect-camtest` and `freenect-glview` to gather more debug information.

JohnnytheRocket, 2016-07-15:
Thanks for your reply!

I ran freenect-camtest and freenect-glview and here's the result:

freenect-camtest:
```
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00
^CShutting down
write_register: 0x0006 <= 0x00
write_register: 0x0005 <= 0x00
Done!
```

And freenect-glview
```
Kinect camera test
Number of devices found: 1
GL thread
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
libGL error: unable to load driver: mali_drm_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: mali_drm
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00
'w' - tilt up, 's' - level, 'x' - tilt down, '0'-'6' - select LED mode, '+' & '-' - change IR intensity 
'f' - change video format, 'm' - mirror video, 'o' - rotate video with accelerometer 
'e' - auto exposure, 'b' - white balance, 'r' - raw color, 'n' - near mode (K4W only) 
 raw acceleration:  -22  714  388  mks acceleration: -0.263426 8.549387 4.645885^C
```

The libGL error in the freenect-glview seems to be because odroid apparently don't support openGL, is that a requisite for libfreenect, or will it only provide issues with some examples?

Other than that I'm not really sure what I'm looking for.

piedar, 2016-07-15:
`freenect-camtest` should be printing a line for every frame it receives.  Maybe something is wrong with your libusb that prevents it from handling isochronous transfers?

JohnnytheRocket, 2016-07-21:
Thank you again for your suggestion. If the libusb is the culprit I have ran out of ideas now, I've tried the last version and the next-to-last version (which stills works on my Raspberry), with both Ubuntu 16.04 and Debian Jessie, as well as with the 0.5.2 version of libfreenect (which is the one the raspberry has) and everything results in the same thing: everything but the depth- and rgb-streaming works.

With freenect-glview I can see the accelerometer data, as well as control the LEDs (I've removed the Kinect's motor, so I'm not sure about the tilting though), and when running I can see the laser turn on, which for the Raspberry always was a surefire way of knowing it works.

Any other ideas to try would be most welcome!

topdjgod, 2017-10-19:
I have the same issue using Kinect v1 on TinkerBoard (Debian OS). No any data be received by using freenect-camtest and freenect-glview. Log is the same as JohnnytheRocket. Any one have an idea?