OpenKinect/libfreenect #476: ssize_t conflicting types in VS2016 x64

armadillu, 2016-07-29:
When using libfreenect inside openFrameworks (wrapped inside ofxKinect) I found that it would not compile under x64 architecture, but it compiles fine under x86. It turns out ssize_t is defined in libusb as __int64 when x64 is the active target.

See the libUSB ssize_t define here: https://github.com/libusb/libusb/blob/master/libusb/libusb.h#L33

The define in question within libfreenect is here: https://github.com/OpenKinect/libfreenect/blob/master/platform/windows/unistd.h#L37

You can see the issue and the patch within OpenFrameowrks here:

https://github.com/openframeworks/openFrameworks/issues/3062#issuecomment-236055201
https://github.com/openframeworks/openFrameworks/pull/5179

armadillu, 2016-07-29:
It's such a small thing that I prepared a PR.

piedar, 2016-07-29:
Thanks for the PR!

closed 2016-07-29 by piedar