OpenKinect/libfreenect #482: Face detection with opencv, open kinect and kinect v2

mindcraft-ixd, 2016-09-12:
hi everyone,

i am trying build a face detection with the opencv and openkinect libraries. for the image input i want to use the xbox kinect v2. i am basing my code on the face detection example within the opencv library. i am working on a mac.

this is my code so far:
<pre>
import gab.opencv.*;
import java.awt.Rectangle;

/* KINECT */
import org.openkinect.freenect.*;
import org.openkinect.freenect2.*;
import org.openkinect.processing.*;

OpenCV opencv;
Kinect2 kinect2;

Rectangle[] faces;

void setup() {
  opencv = new OpenCV(this, 640/2, 480/2);
  size(640, 480);
  // Kinectv2
  kinect2 = new Kinect2(this);
  kinect2.initVideo();
  kinect2.initDevice();
  
  opencv.loadCascade(OpenCV.CASCADE_FRONTALFACE);  
  faces = opencv.detect();
}

void draw() {
  opencv.loadImage(kinect2.getVideoImage());
  image(kinect2.getVideoImage(), 0, 0, 640, 480);

  noFill();
  stroke(0, 255, 0);
  strokeWeight(3);
  for (int i = 0; i < faces.length; i++) {
    rect(faces[i].x, faces[i].y, faces[i].width, faces[i].height);
  }
}
</pre>

the problem seems to be in the line "opencv.loadImage(kinect2.getVideoImage());" since the detection does not work.
when working with the isight camera (using the build-in function "capture" and "video"-add-on) instead of kinect everything works perfectly fine.

can anyone help?

piedar, 2016-09-12:
You should examine the result of kinect2.getVideoImage() to make sure it is valid.
And shouldn't you be calling `opencv.detect()` on every frame instead of just during setup?

mindcraft-ixd, 2016-09-12:
> You should examine the result of kinect2.getVideoImage() to make sure it is valid.

Im not sure what you mean by making "sure it is valid". I can see the video output by declaring "image(kinect2.getVideoImage(), 0, 0, 640, 480);" in the line below. 

I´ve taken a look on at the sourcecode of the openkinect library and "kinect2.getVideoImage()" outputs a pimage if that´s what you´re getting at.

> And shouldn't you be calling opencv.detect() on every frame instead of just during setup?

Since it works perfectly fine while using the isight camera i guess not. But I don´t know though.

yashredevil, 2017-04-18:
How does one use kinect2.getVideoImage() as an input for opencv.loadImage(); ?

yashredevil, 2017-04-18:
I found the solution. 
You need to add this function

void videoEvent (Kinect k) {
  // There has been a video event!
}

closed 2017-05-31 by piedar

idignick, 2017-09-04:
I'm sorry could you elaborate on the "solution" please?