OpenKinect/libfreenect #483: Number of devices found: 0 on OSX

XiaXuehai, 2016-09-26:
`brew update`then,`brew install libfreenect` ,but can't find the devices.

piedar, 2016-09-26:
No idea.  Please provide the output of

    ioreg -p IOUSB -w0

XiaXuehai, 2016-09-26:
```
Pro-2:~ xxh$ ioreg -p IOUSB -w0
+-o Root  <class IORegistryEntry, id 0x100000100, retain 14>
  +-o Root Hub Simulation Simulation@14000000  <class AppleUSBRootHubDevice, id 0x1000002fa, registered, matched, active, busy 0 (2 ms), retain 10>
    +-o Apple Internal Keyboard / Trackpad@14400000  <class AppleUSBDevice, id 0x1000024d3, registered, matched, active, busy 0 (10 ms), retain 19>
    +-o Bluetooth USB Host Controller@14300000  <class AppleUSBDevice, id 0x1000024d9, registered, matched, active, busy 0 (4 ms), retain 21>
    +-o NuiSensor Adaptor      @14200000  <class AppleUSBDevice, id 0x100002ff1, registered, matched, active, busy 0 (1 ms), retain 11>
    +-o NuiSensor Adaptor      @14600000  <class AppleUSBDevice, id 0x100003001, registered, matched, active, busy 0 (1 ms), retain 12>
      +-o Xbox NUI Sensor@14610000  <class AppleUSBDevice, id 0x100003012, registered, matched, active, busy 0 (8 ms), retain 17>
```

piedar, 2016-09-26:
Hmm, that didn't tell me what I thought it would.  What model of Kinect is it?

XiaXuehai, 2016-09-26:
sorry,how to check the model of kinect? I see` model no:1637` on kinect adapter.` model no:1520` on xbox one

piedar, 2016-09-26:
Oh!  If it came with the XBox One, it's a Kinect 2.  This library only supports the first generation.  You'll need https://github.com/OpenKinect/libfreenect2 instead.

closed 2016-09-26 by piedar