OpenKinect/libfreenect #484: Installing Python Wrapper Gives Error

JonnoFTW, 2016-10-06:
Hi,

I'm trying to install the wrapper for python2 only, I did these steps:
```
mkdir build
cd build
cmake -L .. -DCMAKE_INSTALL_PREFIX=/scratch/package/install \
                  -DPYTHON_LIBRARY=/home/CSEM/mack0242/.pyenv/versions/2.7.12/lib/libpython2.7.so \
                  -DPYTHON_INCLUDE_DIR=/home/CSEM/mack0242/.pyenv/versions/2.7.12/include/python2.7  \
                  -DPYTHON2_NUMPY_INCLUDE_PATH=/home/CSEM/mack0242/.pyenv/versions/2.7.12/lib/python2.7/site-packages/numpy/core/include 
make -j8
make install
```
At this point the only contents of `build/wrappers/python/python2` is `freenect.so` and I can't import `freenect` from python. So I tried running the `setup.py` script directly from the root folder (`libfreenect/wrappers/python/setup.py`), and I get the following error:

```
ccache /usr/bin/cc -pthread -shared -L/home/CSEM/mack0242/.pyenv/versions/2.7.12/lib -Wl,-rpath=/home/CSEM/mack0242/.pyenv/versions/2.7.12/lib build/temp.linux-x86_64-2.7/freenect.o -L/home/CSEM/mack0242/.pyenv/versions/2.7.12/lib -R/usr/local/lib -R/usr/local/lib64 -R/usr/lib/ -lusb-1.0 -lfreenect -lfreenect_sync -lpython2.7 -o build/lib.linux-x86_64-2.7/freenect.so
cc: error: unrecognized command line option ‘-R’
cc: error: unrecognized command line option ‘-R’
cc: error: unrecognized command line option ‘-R’
error: command 'ccache' failed with exit status 1
```

This particular error is discussed here: https://github.com/NixOS/nixpkgs/issues/5854 and it looks like the fix is to change `extra_link_args` to `extra_link_args` with `-Wl,rpath` before each of the directories (never mind that this ignores my custom install prefix).

piedar, 2016-10-09:
So the NixOS build system was hiding compiler identity, which defeated gcc-specific flags further down the stack.  Looks like it was [fixed in NixOS/nixpkgs](https://github.com/NixOS/nixpkgs/commit/7dc57a389e3d3eb5b5e8de3b02dfe469496fad1a).  Do you have the latest OS version?

JonnoFTW, 2016-10-09:
I did this build on Ubuntu 14.04 and eventually got it to work.