OpenKinect/libfreenect #487: tilt & led does not work with model 1517

hrask, 2016-11-08:
## I'm using Ubuntu 14.04 with Kinect 1517.
I finished build about freenect with reference to [this page](https://github.com/OpenKinect/libfreenect).
So I can do command `freenect-glview` in terminal. then I can see video and depth.
Several commands were displayed in terminal as follow:
```
'w' - tilt up, 's' - level, 'x' - tilt down, '0'-'6' - select LED mode, '+' & '-' - change IR intensity 
'f' - change video format, 'm' - mirror video, 'o' - rotate video with accelerometer 
'e' - auto exposure, 'b' - white balance, 'r' - raw color, 'n' - near mode (K4W only) 
```
But some command that tilt and led don't work .
I think that other settings are necessary when freenect building.
Next show terminal message.
```
Kinect camera test
Number of devices found: 1
Could not open audio: -1
GL thread
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
[Stream 70] Invalid magic d4ff
[Stream 70] Invalid magic d4ff
[Stream 70] Invalid magic d3ff
[Stream 70] Lost 3 packets
[Stream 70] Lost 3 total packets in 3 frames (1.000000 lppf)
[Stream 70] Invalid magic d4ff
[Stream 70] Invalid magic d4ff
[Stream 70] Invalid magic ffff
[Stream 70] Lost 6 packets
[Stream 70] Lost 9 total packets in 3 frames (3.000000 lppf)
[Stream 70] Lost too many packets, resyncing...
write_register: 0x0047 <= 0x00
```
I saw some solution to similar isues. but they are for Xbox kinect model 1473.
How can I solve for 1517 model?

piedar, 2016-11-08:
For audio support, you must upload firmware to the device. Newer Kinect models may require audio firmware for motor and LED support. The best method is to [insert firmware at runtime](https://github.com/OpenKinect/libfreenect/issues/376#issuecomment-41211251) just after calling `freenect_init()`.

Those files and code I linked will need to be added to any program where you want audio and/or tilt to work.  Unfortunately the headers are C++, so you'll either have to compile the program as C++ or modify the headers to be pure C.

I know it's a PITA and I wish we could just ship the firmware, but the project doesn't want want to risk any legal issues.

hrask, 2016-11-10:
Thank you for answering.
May be your advice is good. 
But I don't know well about C, compile mechanism,and so on.
If you are OK, could you tell me my mistake?

first, I got [2 files.](https://github.com/openframeworks/openFrameworks/tree/master/addons/ofxKinect/src/extra) and put in freenect/include/ .
next, I edited ofxKinectExtras.h like:
```
#pragma once
#define OFX_KINECT_EXTRA_FM

extern unsigned char fwk4wBin[];
int getFWSizek4w(){
     return 145920;
}
unsigned char * getFWDatak4w(){
     return &fwk4wBin[0];
}
```
and remove ofxKinectExtras.cpp after move "unsigned char fwk4wBin[]" of ofxKinectExtras.cpp to libfreenect/examples/glview.c code.

I edited
libfreenect/examples/glview.c row 494
```
if (freenect_init(&f_ctx, NULL) < 0) {
		printf("freenect_init() failed\n");
		return 1;freenect_init
	}
```
to
```
if (freenect_init(&f_ctx, NULL) < 0) {
		printf("freenect_init() failed\n");
		return 1;freenect_init
	}
freenect_set_fw_address_k4w(kinectContext, getFWDatak4w(),getFWSizek4w());
```
I retried to build. but the result did not change.

hrask, 2016-11-10:
I would like to apply this solution to python wrapper, but what should I do it?
thanks.

piedar, 2016-11-13:
There is a way to [load](https://github.com/OpenKinect/libfreenect/blob/master/src/loader.c) the firmware from a file; I know it works with model, 1473, but I never tried it with 1517.  To test it, you'd need to write that byte array you have in source code to a binary file named `audios.bin` and put it somewhere the loader knows to look.

hrask, 2016-11-22:
The model 1517(for windows) seems to be different from model 1473.
so I can't solve. 
Thank you for responding.

metropolik, 2017-05-13:
so uploading new firmware poses no risk of damaging the kinect since the firmware is only temporary?