OpenKinect/libfreenect #488: LIBUSB_ERROR_BUSY

pranali3215, 2016-11-11:
Hi,

When I run freenect-glview, the command only works the first time. 
When I try to run in the second time, it gives the following error.

ubuntu@p:/home/p/Documents/test/# freenect-glview
Kinect camera test
Number of devices found: 1
Failed to claim camera interface: LIBUSB_ERROR_BUSY
Failed to open camera subdevice or it is not disabled.Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.Could not open device

If I unplug the kinect camera and plug it again, it works fine again. But on running it a second time, it gives the same output. 

Is there a way where I don't have to manually unplug and plug it again and again ?

Any help would be appreciated.

piedar, 2016-11-11:
How do you close the window?  Escape key?  And I would not recommend running it as root...

pranali3215, 2016-11-11:
I am actually running it from a python file, which requires root permissions So I have to run it as root. To close it, I kill the python file which calls the process. 

f,res = freenect.sync_get_video()

This is what I call in the python file

piedar, 2016-11-11:
If the process is killed it won't have a chance to clean up.  The python [README says](https://github.com/OpenKinect/libfreenect/blob/ddf092cce09ab4bc97529827a061f1a54a046a17/wrappers/python/README#L18) that it's not necessary to call `sync_stop()` at the end, but I don't see how shutdown would work without it.

You could try using python's [signal](https://docs.python.org/3/library/signal.html) package to handle `SIGINT` and shutdown gracefully [like in camtest.c](https://github.com/OpenKinect/libfreenect/blob/master/examples/camtest.c#L48).  Looks like you'd want to call `freenect.sync_stop()` in the handler.

pranali3215, 2016-11-14:
I have written freenect.sync_stop() in the handler.
It seems to have worked! I do not get this error anymore. 
Thank you for your help

closed 2016-11-14 by pranali3215