OpenKinect/libfreenect #489: Segmentation fault on accessing micropshone

parkerlreed, 2016-11-12:


fhars, 2017-01-07:
I see this too with a 1414 linect (so no firmware required), this is the stacktrace with libfreenect symbols:

```Thread 1 "freenect-micvie" received signal SIGSEGV, Segmentation fault.
0x00007ffff6f152a6 in libusb_submit_transfer () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
(gdb) bt
#0  0x00007ffff6f152a6 in libusb_submit_transfer () from /lib/x86_64-linux-gnu/libusb-1.0.so.0
#1  0x00007ffff7bce9e2 in fnusb_start_iso (dev=0x6555f0, strm=0x655648, cb=0x7ffff7bd0389 <iso_in_callback>, endpoint=130 '\202', xfers=16, pkts=16, len=524) at /home/hars/src/libfreenect/src/usb_libusb10.c:875
#2  0x00007ffff7bd092b in freenect_start_audio (dev=0x655390) at /home/hars/src/libfreenect/src/audio.c:183
#3  0x000000000040219f in main (argc=1, argv=0x7fffffffdd88) at /home/hars/src/libfreenect/examples/micview.c:194
(gdb) up
#1  0x00007ffff7bce9e2 in fnusb_start_iso (dev=0x6555f0, strm=0x655648, cb=0x7ffff7bd0389 <iso_in_callback>, endpoint=130 '\202', xfers=16, pkts=16, len=524) at /home/hars/src/libfreenect/src/usb_libusb10.c:875
875                             int ret = libusb_submit_transfer(strm->xfers[i]);
(gdb) p *strm
$1 = {parent = 0x6555f0, xfers = 0x6464c0, buffer = 0x7ffff7fd5010 "", cb = 0x7ffff7bd0389 <iso_in_callback>, num_xfers = 16, pkts = 16, len = 524, dead = 0, dead_xfers = 0}
(gdb) l
870                     else
871                     {
872                             libusb_fill_iso_transfer(strm->xfers[i], dev->dev, endpoint, bufp, pkts * len, pkts, iso_callback, strm, 0);
873                             libusb_set_iso_packet_lengths(strm->xfers[i], len);
874
875                             int ret = libusb_submit_transfer(strm->xfers[i]);
876                             if (ret < 0)
877                             {
878                                     FN_WARNING("Failed to submit isochronous transfer %d: %d\n", i, ret);
879                                     strm->dead_xfers++;
(gdb) p strm->xfers
$2 = (struct libusb_transfer **) 0x6464c0
(gdb) p strm->xfers[i]
$6 = (struct libusb_transfer *) 0x618348
(gdb) p *strm->xfers[i]
$7 = {dev_handle = 0x0, flags = 0 '\000', endpoint = 130 '\202', type = 1 '\001', timeout = 0, status = LIBUSB_TRANSFER_COMPLETED, length = 8384, actual_length = 0, callback = 0x7ffff7bce46e <iso_callback>, user_data = 0x655648, buffer = 0x7ffff7fd5010 "", num_iso_packets = 16, iso_packet_desc = 0x618384}
```

Here is the reason:

```
Starting program: /usr/local/bin/freenect-micview 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff3fca700 (LWP 22546)]
Number of devices found: 1

Thread 1 "freenect-micvie" hit Breakpoint 3, fnusb_open_subdevices (dev=0x655390, index=0) at /home/hars/src/libfreenect/src/usb_libusb10.c:492
492         if( (ctx->enabled_subdevices & FREENECT_DEVICE_AUDIO) && dev->usb_cam.dev != NULL )
(gdb) p ctx->enabled_subdevices 
$2 = FREENECT_DEVICE_AUDIO
(gdb) p *dev
$3 = {parent = 0x604010, next = 0x0, user_data = 0x0, usb_cam = {parent = 0x655390, dev = 0x0, device_dead = 0, VID = 0, PID = 0}, depth_isoc = {parent = 0x0, xfers = 0x0, buffer = 0x0, cb = 0x0, num_xfers = 0, pkts = 0, len = 0, dead = 0, dead_xfers = 0}, video_isoc = {parent = 0x0, xfers = 0x0, buffer = 0x0, cb = 0x0, num_xfers = 0, pkts = 0, 
    len = 0, dead = 0, dead_xfers = 0}, depth_cb = 0x0, video_cb = 0x0, depth_chunk_cb = 0x0, video_chunk_cb = 0x0, video_format = FREENECT_VIDEO_RGB, depth_format = FREENECT_DEPTH_11BIT, video_resolution = FREENECT_RESOLUTION_LOW, depth_resolution = FREENECT_RESOLUTION_LOW, cam_inited = 0, cam_tag = 0, depth = {running = 0, flag = 0 '\000', 
    synced = 0, seq = 0 '\000', got_pkts = 0, pkt_num = 0, pkts_per_frame = 0, pkt_size = 0, frame_size = 0, last_pkt_size = 0, valid_pkts = 0, lost_pkts = 0, valid_frames = 0, variable_length = 0, last_timestamp = 0, timestamp = 0, split_bufs = 0, lib_buf = 0x0, usr_buf = 0x0, raw_buf = 0x0, proc_buf = 0x0}, video = {running = 0, flag = 0 '\000', 
    synced = 0, seq = 0 '\000', got_pkts = 0, pkt_num = 0, pkts_per_frame = 0, pkt_size = 0, frame_size = 0, last_pkt_size = 0, valid_pkts = 0, lost_pkts = 0, valid_frames = 0, variable_length = 0, last_timestamp = 0, timestamp = 0, split_bufs = 0, lib_buf = 0x0, usr_buf = 0x0, raw_buf = 0x0, proc_buf = 0x0}, registration = {reg_info = {
      dx_center = 0, ax = 0, bx = 0, cx = 0, dx = 0, dx_start = 0, ay = 0, by = 0, cy = 0, dy = 0, dy_start = 0, dx_beta_start = 0, dy_beta_start = 0, rollout_blank = 0, rollout_size = 0, dx_beta_inc = 0, dy_beta_inc = 0, dxdx_start = 0, dxdy_start = 0, dydx_start = 0, dydy_start = 0, dxdxdx_start = 0, dydxdx_start = 0, dxdxdy_start = 0, 
      dydxdy_start = 0, back_comp1 = 0, dydydx_start = 0, back_comp2 = 0, dydydy_start = 0}, reg_pad_info = {start_lines = 0, end_lines = 0, cropping_lines = 0}, zero_plane_info = {dcmos_emitter_dist = 0, dcmos_rcmos_dist = 0, reference_distance = 0, reference_pixel_size = 0}, const_shift = 0, raw_to_mm_shift = 0x0, depth_to_rgb_shift = 0x0, 
    registration_table = 0x0}, usb_audio = {parent = 0x655390, dev = 0x0, device_dead = 0, VID = 0, PID = 0}, audio_out_isoc = {parent = 0x0, xfers = 0x0, buffer = 0x0, cb = 0x0, num_xfers = 0, pkts = 0, len = 0, dead = 0, dead_xfers = 0}, audio_in_isoc = {parent = 0x0, xfers = 0x0, buffer = 0x0, cb = 0x0, num_xfers = 0, pkts = 0, len = 0, dead = 0, 
[... cut ...]
    in_counter = 0 '\000', mic_buffer = {0x0, 0x0, 0x0, 0x0}, cancelled_buffer = 0x0, in_unknown = 0x0}, audio_tag = 0, usb_motor = {parent = 0x655390, dev = 0x0, device_dead = 0, VID = 0, PID = 0}, raw_state = {accelerometer_x = 0, accelerometer_y = 0, accelerometer_z = 0, tilt_angle = 0 '\000', tilt_status = TILT_STATUS_STOPPED}, 
  device_does_motor_control_with_audio = 0, motor_control_with_audio_enabled = 0}
```

The micview.c example never initializes the camera, and so will never discover the audio device.

fhars, 2017-01-07:
Change line 165 of micview.c to

```
        freenect_select_subdevices(f_ctx, (freenect_device_flags)(FREENECT_DEVICE_CAMERA | FREENECT_DEVICE_AUDIO));
```

And the segfault goes away.

parkerlreed, 2017-03-29:
Not sure if related but micview doesn't exit gracefully after making that change (UI freezes and have to kill -9 the process)

parkerlreed, 2017-03-29:
Line 79 of wavrecord.c also has to be changed to avoid the same segfault.

piedar, 2017-06-02:
So far to replicate because freenect-micview and freenect-wavrecord hang immediately with

```
Number of devices found: 1
Failed to submit isochronous transfer 0: -1
Failed to submit isochronous transfer 1: -1
Failed to submit isochronous transfer 2: -1
Failed to submit isochronous transfer 3: -1
Failed to submit isochronous transfer 4: -1
Failed to submit isochronous transfer 5: -1
Failed to submit isochronous transfer 6: -1
Failed to submit isochronous transfer 7: -1
Failed to submit isochronous transfer 8: -1
Failed to submit isochronous transfer 9: -1
Failed to submit isochronous transfer 10: -1
Failed to submit isochronous transfer 11: -1
Failed to submit isochronous transfer 12: -1
Failed to submit isochronous transfer 13: -1
Failed to submit isochronous transfer 14: -1
Failed to submit isochronous transfer 15: -1
```