OpenKinect/libfreenect #491: Can't run setup.py for Python wrappers (Ubuntu 16.04)

iamlegolas, 2016-11-17:
I went through the entire process as stated here on Github:

git clone https://github.com/OpenKinect/libfreenect
cd libfreenect
mkdir build
cd build

cmake .. -DCMAKE_BUILD_PYTHON=ON
make

I then navigate to the folder: /wrappers/python
I try to run the command: python setup.py install
To this, I get the following error: 
/usr/bin/ld: cannot find -lfreenect
/usr/bin/ld: cannot find -lfreenect_sync
collect2: error: ld returned 1 exit status
error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

Needless to say, I can't import freenect in any of my projects.

Please help me out. 
Thanks!

iamlegolas, 2016-11-17:
@JonnoFTW Please take a look

piedar, 2016-11-17:
The python install script assumes that you have already installed the main library into /usr/lib or /usr/local/lib.  To install the main library, execute `sudo make install` from the build directory.  I think it's also possible to run the scripts from the wrappers/python directory without installing anything.

closed 2017-05-31 by piedar

Berlinbenilo, 2022-12-31:
Did the above issue solved?