OpenKinect/libfreenect #494: in flags.c: send_cmd never returns because FLOOD loop always sends full byte count 0x200

cuisinart, 2016-12-19:
I'm not sure where this is going wrong.  Under libusb1, windows 10, visual studio 2015, 
the send_cmd function never returns because it enters an infinite loop flooding the camera with 0x200 bytes of data, expecting to get less than that sent at some point, but it never happens and so the function call never returns.

The kinect is found fine, it just fails during the init when send_cmd is called.  With another build under visual studio 2005, libusb0, this works.  I don't know why the latest libusb doesn't seem to work as the devs expected, but it's strange to me that this infinite loop in flags.c would be allowed when libusb_control_transfer is ale to send all data requested every time.

[code]	do {
		actual_len = fnusb_control(&dev->usb_cam, 0xc0, 0, 0, 0, ibuf, 0x200);
		FN_FLOOD("send_cmd: actual length = %d\n", actual_len);
	} while ((actual_len == 0) || (actual_len == 0x200));
[/code]

cuisinart, 2016-12-28:
Update: changing the build to use libusb0 and libusbemu.cpp, failguard.cpp under platform, and changing from libusb 1.0.dll driver to libsub0.dll driver this now works under visual studio 2015.  There must still be a bug in the libusb subsystem, an incompatibility with libusb1.0 and the windows inf drivers specified in this download, or a problem with this libary's interaction with libusb 1.0 on windows.

piedar, 2016-12-28:
Thank you for reporting this info and workaround!  If you get a chance, please also report it to [libusb](https://github.com/libusb/libusb).