OpenKinect/libfreenect #498: libFreenect + OpenNI

josuerocha, 2017-01-17:
Is it possible to use libFreenect with OpenNI (not OpenNI2)? If so, how?

Thanks a lot.

piedar, 2017-01-17:
There is no way to use original OpenNI with libfreenect at the moment.  But there is a separate Kinect driver  at https://github.com/avin2/SensorKinect.

josuerocha, 2017-01-19:
Thank you for your answer. Do you think the integration with OpenNI2 will be done soon? Because I need to control the motors for the  I am working on.

josuerocha, 2017-01-19:
Also, could anybody use Freenect + OpenNI2 + PCL? Because I could get Freenect + OpenNI2 working perfectly but when I try to manipulste the stream in PCL I only get segmentation faults.

piedar, 2017-01-20:
I have not seen any way to control the motor via OpenNI2.  You could have your application call directly into libfreenect for that, perhaps in a separate helper program.  When using both APIs at once, remember to first call

    freenect_select_subdevices(ctx, FREENECT_DEVICE_MOTOR);

so the helper doesn't try to claim the camera when OpenNI2 is already using it.

closed 2017-05-31 by piedar