OpenKinect/libfreenect #501: Kinect 1414 Could not open device

LewisErick, 2017-04-13:
Hi there! I'm currently using Mac OS X El Capitan 10.11.5 with Kinect Model 1414 to run freenect-glview, however it always gives me this error:

```
Kinect camera test
Number of devices found: 1
Failed to open motor subddevice or it is not disabled.Failed to open audio subdevice or it is not disabled.Could not open device
```

The config I used to build and install it from source is the following:

```
BUILD_AS3_SERVER:BOOL=OFF
BUILD_CPP:BOOL=ON
BUILD_CV:BOOL=OFF
BUILD_C_SYNC:BOOL=ON
BUILD_EXAMPLES:BOOL=ON
BUILD_FAKENECT:BOOL=ON
BUILD_OPENNI2_DRIVER:BOOL=OFF
BUILD_PYTHON:BOOL=OFF
BUILD_PYTHON2:BOOL=OFF
BUILD_PYTHON3:BOOL=OFF
BUILD_REDIST_PACKAGE:BOOL=OFF
CMAKE_BUILD_TYPE:STRING=debug
CMAKE_INSTALL_PREFIX:PATH=/usr/local
CMAKE_OSX_ARCHITECTURES:STRING=
CMAKE_OSX_DEPLOYMENT_TARGET:STRING=
CMAKE_OSX_SYSROOT:STRING=
GLUT_cocoa_LIBRARY:FILEPATH=/System/Library/Frameworks/Cocoa.framework
LIBUSB_1_INCLUDE_DIR:PATH=/usr/local/include/libusb-1.0
LIBUSB_1_LIBRARY:FILEPATH=/usr/local/lib/libusb-1.0.dylib
```

The lsusb gives me the following:
```
Bus 029 Device 010: ID 045e:02ae Microsoft Corporation Xbox NUI Camera  Serial: A00362A12109049A
Bus 029 Device 008: ID 045e:02b0 Microsoft Corporation Xbox NUI Motor 
Bus 029 Device 009: ID 045e:02ad Microsoft Corporation Xbox Kinect Audio, ¬© 2011 Microsoft Corporation. All rights reserved.  Serial: A44882D14248049A
```

piedar, 2017-04-13:
The [code that opens the motor](https://github.com/OpenKinect/libfreenect/blob/4696c803465761158c55f12396a9a1ce1a4b15f5/src/usb_libusb10.c#L438) ought to print out an error if it fails.  But apparently it's not even finding the motor subdevice - maybe `libusb_get_parent()` is not working on your platform?

LewisErick, 2017-04-14:
Indeed, I checked at runtime that the `libusb_get_parent()` function is giving me NULL.
Does that have to do with libusb itself or am I missing something?

`LIBUSB_1_INCLUDE_DIR:PATH=/usr/local/include/libusb-1.0
LIBUSB_1_LIBRARY:FILEPATH=/usr/local/lib/libusb-1.0.dylib`

I installed libusb using Macports like the README said:
`sudo port install git-core cmake libusb libtool`

LewisErick, 2017-04-14:
I've looked more into it and it seems that the parent of the device of the camera is non existent.
Could you please tell me what should the libusb_get_parent(camera) should give me in terms of its description (vendor id and product id).

LewisErick, 2017-04-20:
So at the end I made a workaround which detected the motor and sensor from the camera not by its parent but by the match in the vendor ID. This workaround is not perfect as it does not consider the possibility of two or more Kinects connected at the same time, but that itself is not an issue in what I'm working on. Thanks!

https://github.com/LewisErick/libfreenect/blob/master/src/usb_libusb10.c

FranciscoG, 2017-05-25:
having this exact same `freenect-glview`
I'm on OSX 10.11.6

I'm hoping that this could be resolved without me having to build it locally.

piedar, 2017-05-31:
Fixing incoming!  Please try the latest and reopen if the issue persists.

    brew install --HEAD libfreenect

FranciscoG, 2017-05-31:
brew installed at HEAD and still the same issue :-(

piedar, 2017-06-01:
It required a more careful analysis to fix it properly (2ce5bd36da046d5bdb23b8f83e84b7604f33a9ff).  Please report results!

FranciscoG, 2017-06-02:
works for me now!! thank you!

commit 6943a007 (2017-05-31): Add ability to match orphan audio devices - fixes #501
commit e0fe29ba (2017-05-31): Add ability to match orphan audio devices - fixes #501
closed 2017-07-22 by piedar