OpenKinect/libfreenect #503: OSX error: dyld: Library not loaded: /opt/X11/lib/libglut.3.dylib

FranciscoG, 2017-05-18:
I followed the simple [Homebrew instructions](https://openkinect.org/wiki/Getting_Started#OS_X), making sure I was up to date with Homebrew and all my formulae, and whenever I run `freenect-glview` I get the following error:

```bash
dyld: Library not loaded: /opt/X11/lib/libglut.3.dylib
  Referenced from: /usr/local/bin/freenect-glview
  Reason: image not found
Trace/BPT trap: 5
```
OSX 10.11.6  El Capitan
Macbook Air 11-inch mid 2013

FranciscoG, 2017-05-25:
I have a fix for this specific issue, at least for myself.  Turns out that my X11 directory was located somewhere else.  So I symlinked it:

```bash
$ ln -s /usr/X11 /opt/X11
```
Then I re-ran `freenect-glview` and it worked, sort of.   Got past that original error and it showed me a pop-up that I needed to download and install  XQuartz because the X11 server is no longer part of OSX but is maintained separately at:  https://www.xquartz.org/

and there is where I'm at now.  I'm closing this comment though because I have answered my own original question.  Any other issues I'll open up another issue for.

closed 2017-05-25 by FranciscoG

athorsen, 2017-10-23:
Thank you for this! I ran into the exact same problem, and your solution of installing XQuartz fixed it. Much appreciated!

MichaelTang0115, 2017-11-14:
Same problem using libfreenect on Mac, thanks !

iandiirie, 2020-07-07:
I. had. the.  same error. and installed. Xquartz
but. had other error. that said couldnt. display. image
so. i ran the. freenect-glview.  command on the Xquartz.   app and now is working on
mac mini ios catalina. 15.5
thanks !!

stuwilmur, 2024-03-06:
Brilliant thanks for the fix!