OpenKinect/libfreenect #506: Errors building with Visual Studio 2010 using Windows 10 64 bits and Kinect Xbox 360 Model 1473

cuatrocuatros, 2017-05-27:
Hi everyone.

My operative system is Windows 10 64 bits. I'm using Libusb win32 1.2.6.0, Glut 3.7.6 and Pthreads w32 2-8-0, Kinect Xbox360 model 1473 and Visual Studio 2010. I copied the needed .dll files in system32, which I have as path variable. I followed the Getting Started of Open Kinect (https://openkinect.org/wiki/Getting_Started).  With Cmake I always have all well configured and generated. But then I get errors building with Visual Studio 2010.

So, I have 3 doubts. 

1. As many of you know, with the 1473 version of Kinect I can't see the NUI Motor in the Device Manager of Windows. Should I use Zadig? I downloaded it but I don't know how I have to use it to make the Motor appear. Is it necessary? Also, once I connect the Kinect to my latop, its green light is always blinking, before installing the drivers (through libfreenect) of Nui Camera and Nui Audio and after installing them. Always blinking. Never steady. Is that right?

2. As I have a kinect 1473, I tried with different versions of libfreenect. Building errors change depending on wheter I'm using the general "libfreenect-master" (https://github.com/OpenKinect/libfreenect/tree/master), the "libfreenect-master" version from ofTheo (https://github.com/ofTheo/libfreenect/tree/master), the "bugfix-1473freeze" version from ofTheo (https://github.com/ofTheo/libfreenect/tree/bugfix-1473freeze) or the "feature-k4w-1473-tilt-led" version from ofTheo (https://github.com/ofTheo/libfreenect/tree/feature-k4w-1473-tilt-led). 

First of all, what should I choose as "Current Generator" in Cmake? "Visual Studio 10 2010" or "Visual Studio 10 2010 Win64"?

**Using "Visual Studio 10 2010 Win64" as "Current Generator", once I have all files configured and generated in Cmake, when I try to build the .sln (solution) with Visual Studio in Debug and Win32 (changing 64bit to Win32) I get the same following errors using any of the 4 versions of libfreenect that I described in point 2:**

_Error	1	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\fakenect\fakenect.c	26	1	fakenect
Error	11	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\registration.c	27	1	freenect
Error	12	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\audio.c	26	1	freenect
Error	13	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\loader.c	27	1	freenect
Error	15	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\registration.c	27	1	freenectstatic
Error	16	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\audio.c	26	1	freenectstatic
Error	17	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\loader.c	27	1	freenectstatic
Error	19	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\micview.c	27	1	freenect-micview
Error	20	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\glview.c	32	1	freenect-glview
Error	21	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\regview.c	32	1	freenect-regview
Error	22	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\chunkview.c	32	1	freenect-chunkview
Error	23	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\wavrecord.c	27	1	freenect-wavrecord
Error	24	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\fakenect\record.c	26	1	fakenect-record
Error	25	error C1083: Cannot open include file: 'libfreenect.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\hiview.c	32	1	freenect-hiview
Error	26	error C1083: Cannot open include file: 'stdbool.h': No such file or directory	C:\Program Files (x86)\libfreenect\examples\camtest.c	27	1	freenect-camtest
Error	2	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\core.c	32	1	freenect
Error	3	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\core.c	32	1	freenectstatic
Error	4	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\tilt.c	30	1	freenect
Error	5	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\tilt.c	30	1	freenectstatic
Error	6	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\cameras.c	30	1	freenect
Error	7	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\cameras.c	30	1	freenectstatic
Error	8	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\flags.c	28	1	freenect
Error	9	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\flags.c	28	1	freenectstatic
Error	10	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\usb_libusb10.c	30	1	freenect
Error	14	error C1083: Cannot open include file: 'unistd.h': No such file or directory	C:\Program Files (x86)\libfreenect\src\usb_libusb10.c	30	1	freenectstatic
Error	18	error MSB6003: The specified task executable "CL.exe" could not be run. The process cannot access the file 'C:\Users\lagoa\Desktop\CarpetaCmake\src\Debug\cl.read.1.tlog' because it is being used by another process.	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\Platforms\Win32\Microsoft.Cpp.Win32.Targets	147	6	freenect_

So, after that, I thought I should have used as "**Current Generator**" in Cmake the "**Visual Studio 10 2010**". So I selected it and now depending on which version of libfreenect I'm using, I get the following errors once I build the solution (in Debug and Win32):

**Using the general "libfreenect-master" (https://github.com/OpenKinect/libfreenect/tree/master):**

Solution based in 16 projects. There were in total 592 errors, 155 warnings. I simplified it because it was too long:

_1>------ Build started: Project: ZERO_CHECK, Configuration: Debug Win32 ------
1>  Checking Build System
1>  CMake does not need to re-run because C:/Users/lagoa/Desktop/CarpetaCmake/CMakeFiles/generate.stamp is up-to-date.
1>  CMake does not need to re-run because C:/Users/lagoa/Desktop/CarpetaCmake/src/CMakeFiles/generate.stamp is up-to-date.
1>  CMake does not need to re-run because C:/Users/lagoa/Desktop/CarpetaCmake/examples/CMakeFiles/generate.stamp is up-to-date.
1>  CMake does not need to re-run because C:/Users/lagoa/Desktop/CarpetaCmake/wrappers/c_sync/CMakeFiles/generate.stamp is up-to-date.
2>------ Build started: Project: freenect, Configuration: Debug Win32 ------
3>------ Build started: Project: freenectstatic, Configuration: Debug Win32 ------
4>------ Build started: Project: firmware, Configuration: Debug Win32 ------
5>------ Skipped Build: Project: uninstall, Configuration: Debug Win32 ------
5>Project not selected to build for this solution configuration 
2>  Building Custom Rule C:/Program Files (x86)/libfreenect/src/CMakeLists.txt
2>  CMake does not need to re-run because C:/Users/lagoa/Desktop/CarpetaCmake/src/CMakeFiles/generate.stamp is up-to-date.
2>cl : Command line warning D9002: ignoring unknown option '-O0'
2>cl : Command line warning D9002: ignoring unknown option '-g'
2>  core.c
3>  Building Custom Rule C:/Program Files (x86)/libfreenect/src/CMakeLists.txt
3>  CMake does not need to re-run because C:/Users/lagoa/Desktop/CarpetaCmake/src/CMakeFiles/generate.stamp is up-to-date.
3>cl : Command line warning D9002: ignoring unknown option '-O0'
3>cl : Command line warning D9002: ignoring unknown option '-g'
3>  core.c
[... cut ...]
    freenect_set_fw_address_nui(kinectContext, ofxKinectExtras::getFWData1473(), ofxKinectExtras::getFWSize1473());
    freenect_set_fw_address_k4w(kinectContext, ofxKinectExtras::getFWDatak4w(), ofxKinectExtras::getFWSizek4w());

immediately after freenect_init

you would need to include ofxKinectExtras.h and .cpp and ( which I posted a link above ) and pass in the context to both functions as the first argument."_

What is that? Should I do that? How?

I wasted too much time trying to get libfreenect finally installed. I would appreciate very, very much your help.

Thanks!

piedar, 2017-05-30:
I'm thinking VS 2010 is just too old, but with a little effort I can get it to compile with VS 2015.

cuatrocuatros, 2017-05-30:
No, I didn't solve the errors... Do you think that? Should I try downloading the VS 2015?

piedar, 2017-05-31:
It's worth a try!  A patch already (c29570e).  I must also suggest cloning to your data store instead of Program Files.

cuatrocuatros, 2017-06-09:
Hi, again.

Tried with Visual Studio 2015. Source code in /Documents. And applied the patch: https://github.com/OpenKinect/libfreenect/commit/c29570e393c08305a4bab14ba285a1ae2f2c329c

Same errors :(

closed 2017-06-13 by cuatrocuatros