OpenKinect/libfreenect #509: Can't make fakenect working with python wrappers

rguillome, 2017-06-09:
Hi,

Here is the output I get while trying to launch the python demo_cv_aysnc.py with fakenect. I specified the LD_PRELOAD and FAKENECT_PATH env var :
```shell
Traceback (most recent call last):
  File "demo_cv_async.py", line 2, in <module>
    import freenect
ImportError: /usr/local/lib/libfreenect_sync.so.0.5: undefined symbol: freenect_camera_to_world
```

I investigated a little bit since this [wiki](https://openkinect.org/wiki/Fakenect#Evaluation) page presents a sample with this python demo, it should be my mistake ...

So I dig into the freenect import build from C libs with cython : the libfreenect_sync.h (and so on libfreenect_sync.c) declares a ```freenect_sync_camera_to_world``` and its implementation makes a call to freenect_camera_to_world. But fakenect.c which should fake the call to the real kinect I/O doesn't implements this method but moreover, the libfreenect.h doesn't declares this signature...

Why this method isn't overrided or just cloned under the fakenect.c source code?
I will try a fork but It should be nice to have your return BTW.

Regards,
Guillaume

piedar, 2017-06-12:
The method `freenect_sync_camera_to_world()` is declared in freenect_registration.h, but your observation is correct: it has not yet been implemented in fakenect.

rguillome, 2017-06-12:
I dig deeper and found that my first analysis was Wrong. Indeed, a patch has already been made adding the header lib_registration.h to the c_sync wrappers which is the synchronous bridge between python wrappers and the libfreenect (which by design is asynchronous).
With the LD_PRELOAD env var, the fakenect resulting libs only overrides a couple of target functions that make real call to the device. So `freenect_camera_to_world` should be offered by the usual libfreenect.so. 

Two tests that shows that having the same library name between the fakenect one and the usual one is (now?), a mistake :

I finally expected a mistake in the librairies links. So

rguillome, 2017-06-13:
This issue is corrected by the Pull request : #511

closed 2017-06-13 by rguillome