OpenKinect/libfreenect #510: sync_get_depth() getting stuck and never returns

fabiocapsouza, 2017-06-12:
I've been using the Kinects with Raspberries Pi 3 in a project for a while and recently I've been encontering some problems where the function **sync_get_depth()** (of the python wrapper) writes some errors messages in stderr but never returns, leaving my program stuck forever. This doesn't happen very often, but I couldn't find a way to recover from it without restarting the program or rebooting. I tried reseting the USB bus using [this approach](https://raspberrypi.stackexchange.com/questions/6782/commands-to-simulate-removing-and-re-inserting-a-usb-peripheral) but it didn't help. I don't have physical access to the Raspberries nor the Kinects, but I can access it via SSH.

I install the library using `sudo apt-get install python-freenect`.

Stderr's output is:
> send_cmd: Input control transfer failed (12)
> send_cmd() returned 12 [0000 0000], 0000 expected
> send_cmd: Output control transfer failed (-1)
> USB device disappeared, cancelling stream 82 :(
> USB camera marked dead, stopping streams
> send_cmd: Output control transfer failed (-4)

Any help would be appreciated.

piedar, 2017-06-13:
It probably gets stuck waiting forever on `cb_cond`, which is never fired because the frames stop coming.

ppgiannak, 2019-06-14:
I have the same problem on Ubuntu 16.04. 
freenect-glview works fine but sync_get_depth() hangs forever on python 3.5.