OpenKinect/libfreenect #512: Number of devices found: 0

arlequin99, 2017-06-21:
I've tried to run freenect-glview but don't run.
Only this message appears

Kinect camera test
Number of devices found: 0

I did install it by

git clone https://github.com/OpenKinect/libfreenect
cd libfreenect
mkdir build
cd build
cmake .. 
make
make install 

Try running as root but it didn't work

I am using opensuse openSUSE Tumbleweed 42.2 

dmesg -Hw

`[  +0.000005] usb 1-2.1: Product: Xbox Kinect Audio, © 2011 Microsoft Corporation. All rights reserved.
[  +0.000003] usb 1-2.1: Manufacturer: Microsoft
[  +0.000004] usb 1-2.1: SerialNumber: A44883C00582131A
[  +0.602595] usb 1-2.1: USB disconnect, device number 13
`

piedar, 2017-06-23:
Does the USB disconnect happen before running freenect-glview?

arlequin99, 2017-06-23:
No, sorry.
I think that the drive is installed correctly  

[ +0.000005] usb 1-2.1: Product: Xbox Kinect Audio, © 2011 Microsoft Corporation. All rights reserved.