OpenKinect/libfreenect #515: Raw depth information in Python wrapper

arsh09, 2017-07-01:
Hello, 

I recently compiled Libfreenect on Ubuntu 14.04 with Python2 wrapper. I would like to detect a ball (using OpenCV) and after detection, I would like to convert the ball's pixels into its world coordinates in mm or meters. 
I looked at the freenect.sync_get_depth() function which I believe it returns values in 0-255 np.array. I might be missing a conversion here (I looked at Kinect Imaging Information guide). If someone can tell me how can I access the raw depth data either from the depth image or from any direct function, it would be amazing. 

Thank you.

arsh09, 2017-07-13:
Sorry to bother. I should have read the files more carefully. For anyone who needs to get the depth image (there are different formats and you can get them by passing the arguments). 

This helped: 

https://stackoverflow.com/questions/23901220/how-do-i-get-kinect-depth-image-data-in-centimeters-using-the-libfreenect-python

Thank you again for the library. it is a wonderful library.

piedar, 2017-07-13:
Sorry I forgot about this question.  Glad you found the answer!

closed 2017-07-13 by piedar