OpenKinect/libfreenect #520: Mac OSX Sierra, Python.

MattTucker22689, 2017-07-26:
Ok, so I ran thru "brew install libfreenect, " as instructed in the ReadMe. But when I go to import it, IDLE keeps giving me this message:

"Traceback (most recent call last):
  File "<pyshell#0>", line 1, in <module>
    import freenect
ModuleNotFoundError: No module named 'freenect'"

I'm running OSX Sierra.
Help?

piedar, 2017-07-26:
If you install python via brew, it should just work.  Otherwise, [add the brew directory to `PYTHONPATH`](https://stackoverflow.com/q/17033096/1925996).

MattTucker22689, 2017-07-27:
If I use "brew link --overwrite python" will I need to worry about not being able to use Numpy, OpenCV, or anything of the sort? And could that fix the problem?

Also, I'm not 100% clear on what to do with "export PYTHONPATH=`brew --prefix`/lib/python2.7/site-packages:$PYTHONPATH." When I entered back into "~/.bash_profile" via ''nano", I saw NO changes? What am I not catching?

MattTucker22689, 2017-08-02:
Thoughts? Further clarification...?

piedar, 2017-08-02:
(I'm assuming python2)  If you type this in the terminal

 ```
export PYTHONPATH=`brew --prefix`/lib/python2.7/site-packages:$PYTHONPATH
echo $PYTHONPATH
```

you should see that the brew directory has been added to `PYTHONPATH`.  In another window, go to that site-packages directory and confirm there's a freenect module there.  Then

```
idle
>>> import freenect
```

If that doesn't work, I'll have to track down a Mac and try it myself.

MattTucker22689, 2017-08-03:
So, I went back and tried it again... Here is everything straight from terminal:

[image]

MattTucker22689, 2017-08-07:
So, I've done some more digging... Still no dice... But there's this:

[image]

So I know it downloaded. But that's literally where it stops... Because nothing wants to work

MattTucker22689, 2017-08-07:
So, I just downloaded X11, and can now run freenect-glview from terminal.... but still can't import freenect in Python. Also, yes I'm using Python 2.7.

[image]

piedar, 2017-08-07:
D'oh, I didn't realize that the [Formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/libfreenect.rb) doesn't build any python bits.  This manual way worked for me.

```
brew install cython

git clone https://github.com/OpenKinect/libfreenect
cd libfreenect
mkdir build
cd build
cmake .. -DBUILD_PYTHON2=ON -DCYTHON_EXECUTABLE="/usr/local/opt/cython/bin/cython"
make

cd wrappers/python/python2
python
>>> import freenect
```

The permanent solution will be to patch the brew Formula to take an optional dependency on python and cython.

MattTucker22689, 2017-08-08:
SOLID! IT WORKS NOW!

Thank you so very much for your help!

MattTucker22689, 2017-08-22:
So how does one "patch the Formula to take an optional dependency on python and cython" for the permanent fix?

piedar, 2017-08-22:
I'm trying to do that at [homebrew-core#16772](https://github.com/Homebrew/homebrew-core/pull/16772), but there are some remaining issues with the build process.

eyun89, 2018-12-12:
@piedar 
Hi piedar, 

Do you happen to have a solution for python 3.7?

I pretty much followed the instructions on the readme and I'm stuck on the step where i type `python setup.py install` in the `libfreenect/wrappers/python` directory. I have no where else to look for help...

This is my output:
```
running install
running build
running build_ext
skipping 'freenect.c' Cython extension (up-to-date)
building 'freenect' extension
gcc -Wno-unused-result -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -I/Users/eyun89/anaconda3/include -arch x86_64 -I/Users/eyun89/anaconda3/include -arch x86_64 -I/Users/eyun89/anaconda3/include/python3.7m -c freenect.c -o build/temp.macosx-10.7-x86_64-3.7/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /Users/eyun89/anaconda3/lib/python3.7/site-packages/numpy/core/include
In file included from /Users/eyun89/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1823:0,
                 from /Users/eyun89/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:18,
                 from /Users/eyun89/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from freenect.c:353:
/Users/eyun89/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:15:2: warning: #warning "Using deprecated NumPy API, disable it by " "#defining NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it by " \
  ^
In file included from freenect.c:355:0:
../../include/libfreenect.h:567:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
 FREENECTAPI int freenect_get_video_mode_count();
 ^
../../include/libfreenect.h:619:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
 FREENECTAPI int freenect_get_depth_mode_count();
 ^
freenect.c: In function ‘__Pyx_ExceptionSave’:
freenect.c:11588:19: error: ‘PyThreadState’ has no member named ‘exc_type’
     *type = tstate->exc_type;
                   ^
freenect.c:11589:20: error: ‘PyThreadState’ has no member named ‘exc_value’
     *value = tstate->exc_value;
                    ^
freenect.c:11590:17: error: ‘PyThreadState’ has no member named ‘exc_traceback’
     *tb = tstate->exc_traceback;
                 ^
freenect.c: In function ‘__Pyx_ExceptionReset’:
freenect.c:11602:22: error: ‘PyThreadState’ has no member named ‘exc_type’
     tmp_type = tstate->exc_type;
                      ^
freenect.c:11603:23: error: ‘PyThreadState’ has no member named ‘exc_value’
     tmp_value = tstate->exc_value;
                       ^
freenect.c:11604:20: error: ‘PyThreadState’ has no member named ‘exc_traceback’
     tmp_tb = tstate->exc_traceback;
                    ^
freenect.c:11605:11: error: ‘PyThreadState’ has no member named ‘exc_type’
     tstate->exc_type = type;
           ^
freenect.c:11606:11: error: ‘PyThreadState’ has no member named ‘exc_value’
     tstate->exc_value = value;
           ^
freenect.c:11607:11: error: ‘PyThreadState’ has no member named ‘exc_traceback’
     tstate->exc_traceback = tb;
           ^
error: command 'gcc' failed with exit status 1
```

Thanks in advanced for your time.

RenFinkle, 2019-07-06:
Experiencing the same behavior as @eyun89 with a slightly different message:
```
python setup.py install
running install
running build
running build_ext
skipping 'freenect.c' Cython extension (up-to-date)
building 'freenect' extension
gcc -pthread -B /home/ren/anaconda3/compiler_compat -Wl,--sysroot=/ -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -Wstrict-prototypes -fPIC -I/home/ren/anaconda3/include/python3.7m -c freenect.c -o build/temp.linux-x86_64-3.7/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /home/ren/anaconda3/lib/python3.7/site-packages/numpy/core/include
In file included from /home/ren/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1822:0,
                 from /home/ren/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /home/ren/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from freenect.c:353:
/home/ren/anaconda3/lib/python3.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
 #warning "Using deprecated NumPy API, disable it with " \
  ^~~~~~~
In file included from freenect.c:355:0:
../../include/libfreenect.h:567:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
 FREENECTAPI int freenect_get_video_mode_count();
 ^~~~~~~~~~~
../../include/libfreenect.h:619:1: warning: function declaration isn’t a prototype [-Wstrict-prototypes]
 FREENECTAPI int freenect_get_depth_mode_count();
 ^~~~~~~~~~~
freenect.c: In function ‘__Pyx_ExceptionSave’:
freenect.c:11588:21: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     *type = tstate->exc_type;
                     ^~~~~~~~
                     curexc_type
freenect.c:11589:22: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     *value = tstate->exc_value;
                      ^~~~~~~~~
                      curexc_value
freenect.c:11590:19: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     *tb = tstate->exc_traceback;
                   ^~~~~~~~~~~~~
                   curexc_traceback
freenect.c: In function ‘__Pyx_ExceptionReset’:
freenect.c:11602:24: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     tmp_type = tstate->exc_type;
                        ^~~~~~~~
                        curexc_type
freenect.c:11603:25: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     tmp_value = tstate->exc_value;
                         ^~~~~~~~~
                         curexc_value
freenect.c:11604:22: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     tmp_tb = tstate->exc_traceback;
                      ^~~~~~~~~~~~~
                      curexc_traceback
freenect.c:11605:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_type’; did you mean ‘curexc_type’?
     tstate->exc_type = type;
             ^~~~~~~~
             curexc_type
freenect.c:11606:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_value’; did you mean ‘curexc_value’?
     tstate->exc_value = value;
             ^~~~~~~~~
             curexc_value
freenect.c:11607:13: error: ‘PyThreadState {aka struct _ts}’ has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
     tstate->exc_traceback = tb;
             ^~~~~~~~~~~~~
             curexc_traceback
error: command 'gcc' failed with exit status 1
```

piedar, 2019-07-08:
Hmm, maybe you [need a newer version of cython](https://stackoverflow.com/questions/51963619/pyproj-fails-to-compile-when-i-pip-install-it-and-its-not-about-gcc)?

RenFinkle, 2019-07-08:
@piedar Perfect! That did it

seanngpack, 2019-10-13:
Any idea how to fix this when using virtualenv? Openfreenect compiles fine--I can view the C++ demos just fine. And I also have the latest Cython in my virtualenv.

<details open>
<summary>libs in my virtualenv</summary>
<br>

```python
Package         Version
--------------- -------
cycler          0.10.0 
Cython          0.29.13
kiwisolver      1.1.0  
matplotlib      3.1.1  
numpy           1.17.2 
pip             19.2.3 
pyparsing       2.4.2  
python-dateutil 2.8.0  
setuptools      41.4.0 
six             1.12.0 
wheel           0.33.6 
```
</details>

<details open>
<summary>error</summary>
<br>

```python

running install
running build
running build_ext
skipping 'freenect.c' Cython extension (up-to-date)
building 'freenect' extension
clang -Wno-unused-result -Wsign-compare -Wunreachable-code -fno-common -dynamic -DNDEBUG -g -fwrapv -O3 -Wall -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/Tk.framework/Versions/8.5/Headers -I/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m -c freenect.c -o build/temp.macosx-10.15-x86_64-3.7/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /Users/seanngpack/.local/share/virtualenvs/cv/lib/python3.7/site-packages/numpy/core/include
In file included from freenect.c:353:
In file included from /Users/seanngpack/.local/share/virtualenvs/cv/lib/python3.7/site-packages/numpy/core/include/numpy/arrayobject.h:4:
In file included from /Users/seanngpack/.local/share/virtualenvs/cv/lib/python3.7/site-packages/numpy/core/include/numpy/ndarrayobject.h:12:
In file included from /Users/seanngpack/.local/share/virtualenvs/cv/lib/python3.7/site-packages/numpy/core/include/numpy/ndarraytypes.h:1830:
/Users/seanngpack/.local/share/virtualenvs/cv/lib/python3.7/site-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: 
      "Using deprecated NumPy API, disable it with "          "#define NPY_NO_DEPRECATED_API
      NPY_1_7_API_VERSION" [-W#warnings]
#warning "Using deprecated NumPy API, disable it with " \
 ^
freenect.c:8444:134: warning: code will never be executed [-Wunreachable-code]
  ...__pyx_t_2++; if (unlikely(0 < 0)) {__pyx_filename = __pyx_f[1]; __pyx_lineno = 794; __...
                                                         ^~~~~~~
freenect.c:11588:21: error: no member named 'exc_type' in 'struct _ts'
    *type = tstate->exc_type;
            ~~~~~~  ^
freenect.c:11589:22: error: no member named 'exc_value' in 'struct _ts'; did you mean
      'curexc_value'?
    *value = tstate->exc_value;
                     ^~~~~~~~~
                     curexc_value
/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/pystate.h:240:15: note: 
      'curexc_value' declared here
    PyObject *curexc_value;
              ^
freenect.c:11590:19: error: no member named 'exc_traceback' in 'struct _ts'; did you mean
      'curexc_traceback'?
    *tb = tstate->exc_traceback;
                  ^~~~~~~~~~~~~
                  curexc_traceback
/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/pystate.h:241:15: note: 
      'curexc_traceback' declared here
    PyObject *curexc_traceback;
              ^
freenect.c:11602:24: error: no member named 'exc_type' in 'struct _ts'
    tmp_type = tstate->exc_type;
               ~~~~~~  ^
freenect.c:11603:25: error: no member named 'exc_value' in 'struct _ts'; did you mean
      'curexc_value'?
    tmp_value = tstate->exc_value;
                        ^~~~~~~~~
                        curexc_value
/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/pystate.h:240:15: note: 
      'curexc_value' declared here
    PyObject *curexc_value;
              ^
freenect.c:11604:22: error: no member named 'exc_traceback' in 'struct _ts'; did you mean
      'curexc_traceback'?
    tmp_tb = tstate->exc_traceback;
                     ^~~~~~~~~~~~~
                     curexc_traceback
/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/pystate.h:241:15: note: 
      'curexc_traceback' declared here
    PyObject *curexc_traceback;
              ^
freenect.c:11605:13: error: no member named 'exc_type' in 'struct _ts'
    tstate->exc_type = type;
    ~~~~~~  ^
freenect.c:11606:13: error: no member named 'exc_value' in 'struct _ts'; did you mean
      'curexc_value'?
    tstate->exc_value = value;
            ^~~~~~~~~
            curexc_value
/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/pystate.h:240:15: note: 
      'curexc_value' declared here
    PyObject *curexc_value;
              ^
freenect.c:11607:13: error: no member named 'exc_traceback' in 'struct _ts'; did you mean
      'curexc_traceback'?
    tstate->exc_traceback = tb;
            ^~~~~~~~~~~~~
            curexc_traceback
/usr/local/Cellar/python/3.7.4_1/Frameworks/Python.framework/Versions/3.7/include/python3.7m/pystate.h:241:15: note: 
      'curexc_traceback' declared here
    PyObject *curexc_traceback;
              ^
2 warnings and 9 errors generated.
error: command 'clang' failed with exit status 1
```
</details>

seanngpack, 2019-10-13:
Okay, I think I fixed the issue. I had to follow the errors and go into freenect.c and change each call to exc_value, exc_traceback, exc_type to curexc_value, curexc_traceback, curexc_type.

dvbit, 2019-11-09:
Great! I have the same error. I will try it

Obrepal, 2021-11-02:
> D'oh, I didn't realize that the [Formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/libfreenect.rb) doesn't build any python bits. This manual way worked for me.
> 
> ```
> brew install cython
> 
> git clone https://github.com/OpenKinect/libfreenect
> cd libfreenect
> mkdir build
> cd build
> cmake .. -DBUILD_PYTHON2=ON -DCYTHON_EXECUTABLE="/usr/local/opt/cython/bin/cython"
> make
> 
> cd wrappers/python/python2
> python
> >>> import freenect
> ```
> 
> The permanent solution will be to patch the Formula to take an optional dependency on python and cython.

I did the same but still after running python3 -c'import freenect' I have ModuleNotFoundError: No module named 'freenect'. I feel hopeless also I am beginner so many things are new for me.

matheusft, 2023-01-29:
Can someone give me a hand here?

Installing libfreenect using homebrew work fine for me using:

`brew install libfreenect`

but apparently, it comes with no wrappers?
Trying to import `freenect` from python does not work as well...

Where do I go from here?
[image]

Wissiak, 2023-03-10:
Same for me

> macOS Ventura 13.1 Python 3.8
> 
> Can someone give me a hand here?
> 
> Installing libfreenect using homebrew work fine for me using:
> 
> `brew install libfreenect`
> 
> but apparently, it comes with no wrappers? Trying to import `freenect` from python does not work as well... Running `/opt/homebrew/Cellar/libfreenect/0.6.4/bin/freenect-glview` works fine for me.
> 
> [image]
> 
> Where do I go from here for importing the lib in python? [image]

Wissiak, 2023-03-10:
> D'oh, I didn't realize that the [Formula](https://github.com/Homebrew/homebrew-core/blob/master/Formula/libfreenect.rb) doesn't build any python bits. This manual way worked for me.
>
> ```
> brew [...]

I tried this (-DCYTHON_EXECUTABLE=/opt/homebrew/Cellar/cython/0.29.33/bin/cython worked for me) but then I got the following error:
[image]

@matheusft I suggest you try the steps mentioned as well and see what happens.