OpenKinect/libfreenect #522: Multiple kinects. Holes on overlapped depth images

Negvet, 2017-07-30:
I am using multiple kinects to increase depth FOV using
```
def get_depth():
    array,_ = freenect.sync_get_depth(kinect_id)
    array = array.astype(np.uint8)
    return array
```

I need to have some overlapping area between two kinects. But, on the overlapping area, there are holes on the depth images.

There are some solutions like http://www.3dsa.kr/fullpapers/2014-021.pdf . 
But I am interested is it possible using libfreenect just to grab depth images one by one from multiple kinects, just by turning kinects on and off in certain sequence (or at least turning on and off just LED). The depth images grabbing are getting shifted in time for multiple kinects in this manner. Have to say that inserting a delay between get_depth1() and get_depth2() does not affect holes, because it looks like LEDs are flashing all the time, in spite of grabbing new frames by libfreenect.
Let's say that I do not need to get 30 fps, 10-15 is enough for me. I am using python wrapper.

Thanks a lot for your work!

piedar, 2017-07-30:
The IR projector is controlled by the depth stream, so the first thing to try is to constantly cycle it.

```
while isRunning:
  depth1 = get_depth(device1)
  freenect.stop_depth(device1)
  # might need to sleep here, but will ruin FPS
  depth2 = get_depth(device2)
  freenect.stop_depth(device2)
```

However, I don't know if that code above will work nicely with the sync wrapper, so maybe you would end up with something more like [demo_mp_async.py](https://github.com/OpenKinect/libfreenect/blob/master/wrappers/python/demo_mp_async.py) with some custom logic in the depth callback.

```
def on_depth_frame(device, data, timestamp):
  freenect.stop_depth(device)

  # do something with the data...

  # might need to sleep here, but will ruin FPS
  other_device = device2 if device == device1 else device2
  freenect.start_depth(other_device)  
```

Negvet, 2017-07-31:
in `freenect.stop_depth(device1`) `device1` have to be `freenect.DevPtr`
Just inserting integer do not work:

> TypeError: Argument 'dev' has incorrect type (expected freenect.DevPtr, got int)

How to get `freenect.DevPtr` of my devices in python?

piedar, 2017-07-31:
Ah, you have to do

```
context = freenect.init()
device1 = freenect.open_device(context, 0)
device2 = freenect.open_device(context, 1)
```

Once again, I don't know if that will play nice with the sync methods.

Negvet, 2017-07-31:
Unfortunately, it reduced FPS down to 1. 

I tested it just with one kinect:
```
while(1):    
    img = get_depth()
    freenect.stop_depth(device1)
    # display the img
    # proceed without delay
    freenect.start_depth(device1)
```

And IR led really turn on and off at every iteration.
Thanks!

closed 2017-07-31 by Negvet