OpenKinect/libfreenect #524: kinect error with raspberyr pi

iman67, 2017-08-16:
I have a problem with use Kinect on a Raspberry Pi. when I run my program with libfreenect, I have an error to run the program, after then I execute more again and more again and suddenly system has work and my kinect wake-up and runs my program. please help. my error is :

 send_cmd: Input control transfer failed (18) freenect_fetch_reg_const_shift: send_cmd read 18 bytes (expected 8) freenect_camera_init(): Failed to fetch const shift for device Error: Invalid index [0] Error: Can't open device. 1.) is it plugged in? 2.) Read the README Traceback (most recent call last):   File "9605019hsn2.py", line 64, in <module>     depth = getDepthMap()   File "9605019hsn2.py", line 42, in getDepthMap     depth, timestamp = freenect.sync_get_depth() TypeError: 'NoneType' object is not iterable

aristizabal95, 2017-12-18:
I also have this error on my raspberry pi with kali linux.

iman67, 2017-12-19:
I solved this problem,
this problem caused by low voltage USB raspberry pi.
you should ensure your raspberry pis voltage isn't low.
actually, you should reconnect your Kinect USB cable to reset Kinect for the run.

aristizabal95, 2017-12-19:
Thanks for answering, is your kinect now working without a problem? I realized yesterday about the low voltage, and now it's a little more stable. Yet, in order to make a program run I must run the program multiple times until it works. Meaning that my kinect is still getting disconnected for some reason

AIWintermuteAI, 2020-05-10:
It's an old issue, but just in case anybody encounters it, in search of solutions - I recently wrote an article about using Kinect 360 and RTAB-MAP ROS with Rasberry Pi 4, with some tweaks it does work now!
https://www.hackster.io/dmitrywat/rgb-d-slam-with-kinect-on-raspberry-pi-4-ros-melodic-ace795

amm98d, 2020-12-12:
> ```
> send_cmd: Input control transfer failed (18)
> freenect_fetch_reg_const_shift: send_cmd read 18 bytes (expected 8)
> freenect_camera_init(): Failed to fetch const shift for device
> Error: Invalid [...]

I have the same error when I run the python scripts inside wrappers with Python3. They run fine with Python 2.7 but not with Python 3. They give this error with Python3. And the rest of my project is in Python3. And there isn't any under voltage issue either. I have checked with dmesg. Please help.