OpenKinect/libfreenect #526: Kinnect 1414 (AKA Old) for 360, do not work

neoxharsh, 2017-10-09:
I tried using the library, everything else works, I am able to change the led, change tilt, but the video feed is not showing up at all. It shows invalid magic. 

So I opened it up, and disconnected the IR projector and then tried running the example, no I am able to get the raw camera feed without problem, but as the ir projector is not connected, depth sensing do not work. Connecting the IR projector back, again gives me the same error. No more stream, but sometimes, i do get depth image, but that only for a second and then it freezes. Can anyone help me out?

piedar, 2017-10-09:
What kind of host computer?  What is the output of `freenect-camtest`?

rib, 2017-11-07:
On a related note I can say that I recently started playing with libfreenect with a Kinect 1414 and also had a similar sounding issue with no depth/video capture but still able to change the tilt and led etc. I was getting lots of URB size mismatch warnings from the kernel in my case running with the generic ArchLinux 4.13.0 kernel. At some point I rebuilt a minimal vanilla 4.13.0 kernel since I was preparing to dig in to figure out really what was broken but luckily for me things started to work on that minimal kernel build. Since it's working for me now I haven't had cause to delve deeper but I assume now there's a particular module that's causing a conflict. I did try unloading+blacklisting the gspca kinect driver as a seemingly likely candidate but that didn't seem to be the culprit for me.

Bulwagga, 2017-11-27:
I  have the same issue.  Kinect 1414.  Running Ubuntu 17.10, tried both X and Wayland.  I used to run this Kinect with the Faro software until an upgrade a few months ago, so I know it's a working device.

freenect-camtest gives me 
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00
 and then hangs.  So I hit control C and then I get
^CShutting down
write_register: 0x0006 <= 0x00
write_register: 0x0005 <= 0x00
Done!

blobfish, 2018-09-02:
I am having problems with the 1414 xbox 360 device also. Depth is working good, but for the rgb video I get, maybe a frame every minute. I have the kernel driver blacklisted on debian testing(buster). Here is the start of camtest.

freenect-camtest
[Stream 70] Negotiated packet size 1920
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00
[Stream 80] Lost 1 total packets in 0 frames (inf lppf)
[Stream 80] Lost 8 total packets in 0 frames (inf lppf)
Received depth frame at 1052486099
[Stream 80] Lost 9 total packets in 0 frames (inf lppf)
[Stream 80] Lost 10 total packets in 0 frames (inf lppf)
[Stream 80] Lost 14 total packets in 0 frames (inf lppf)
[Stream 80] Lost 20 total packets in 0 frames (inf lppf)
Received depth frame at 1054488254
[Stream 80] Lost 22 total packets in 0 frames (inf lppf)
[Stream 80] Lost 25 total packets in 0 frames (inf lppf)
[Stream 80] Lost 26 total packets in 0 frames (inf lppf)
[Stream 80] Lost 29 total packets in 0 frames (inf lppf)
Received depth frame at 1056490409
[Stream 80] Lost 30 total packets in 0 frames (inf lppf)
[Stream 80] Lost 36 total packets in 0 frames (inf lppf)
[Stream 80] Lost 37 total packets in 0 frames (inf lppf)
[Stream 80] Lost 42 total packets in 0 frames (inf lppf)
[Stream 80] Lost 44 total packets in 0 frames (inf lppf)
[Stream 80] Lost 47 total packets in 0 frames (inf lppf)
[Stream 80] Lost 48 total packets in 0 frames (inf lppf)
[Stream 80] Lost 52 total packets in 0 frames (inf lppf)
Received depth frame at 1058492564
[Stream 80] Lost 54 total packets in 0 frames (inf lppf)
[Stream 80] Lost 60 total packets in 0 frames (inf lppf)
[Stream 80] Lost 61 total packets in 0 frames (inf lppf)
[Stream 80] Lost 63 total packets in 0 frames (inf lppf)
[Stream 80] Lost 64 total packets in 0 frames (inf lppf)
[Stream 80] Lost 68 total packets in 0 frames (inf lppf)
...

GmGniap, 2018-09-28:
I'm facing like this lost packets with my Kinect 1414, ros kinetic . Is there any other solutions/driver to work with 1414?

emdete, 2020-01-04:
it seems i ran in the same problem when trying to switch to py3. i compiles HEAD and while the debian packages worked fine (v0.5.3) HEAD did not but showed 

```
[Stream 80] Invalid magic 506c
[Stream 80] Invalid magic 320c
[Stream 80] Lost too many packets, resyncing...
```

I checked which version introduced this behaviour and found 4696c803465761158c55f12396a9a1ce1a4b15f5 to be the commit to blame. the version before (it is 873a2cc6f625221d34856c6574a959cfe8277a1e) worked fine.

it seems the 4696c80 ("Add notices about Kinect v2 Fixes #485", 2016-10-23) broke it with the given behaviour.

FluffyLL, 2023-12-23:
I also have this problem, kinect 1414, on windows 10
```
I connect the kinect, run the program and everything works, sometimes these errors appear randomly:
[Stream 70] Lost too many packets, resyncing...
[Stream 80] Invalid magic 2719
[Stream 80] Inconsistent flag 81 with 122 packets in buf (162 total), resyncing...
[Stream 70] Invalid magic ceff
[Stream 70] Invalid magic aeff
[Stream 70] Invalid magic afff
[Stream 70] Lost too many packets, resyncing...
[Stream 80] Invalid magic 050a
[Stream 80] Inconsistent flag 85 with 75 packets in buf (162 total), resyncing...
[Stream 70] Invalid magic b5ff
[Stream 70] Invalid magic b5ff
[Stream 70] Lost too many packets, resyncing...
[Stream 80] Invalid magic 9598
[Stream 80] Inconsistent flag 85 with 128 packets in buf (162 total), resyncing...
[Stream 70] Invalid magic 70ee
[Stream 80] Invalid magic 3a1d
[Stream 70] Invalid magic d7ff
[Stream 80] Invalid magic 291c
[Stream 70] Inconsistent flag 75 with 194 packets in buf (242 total), resyncing...
[Stream 80] Inconsistent flag 81 with 100 packets in buf (162 total), resyncing...
Got cancelled transfer, but we didn't request it - device disconnected?
USB camera marked dead, stopping streams
[Stream 70] Invalid magic daff
[Stream 70] Invalid magic daff
[Stream 70] Lost too many packets, resyncing...
[Stream 70] Invalid magic aaff
[Stream 70] Invalid magic aaff
```

sometimes freenect_process_events() throws an error and communication stops.
In any case, when If i close and reopen the program I get the error:
`Isochronous transfer error: 1`

And only tilt, led and accelerometer works, no video callback or buffer data received
unplugging and plugging back the usb or power source of the kinect before opening the program fixes it though