OpenKinect/libfreenect #532: upload_firmware: failed to find firmware file. upload_firmware failed: -2

yavier123, 2017-11-17:
I Running the libfreenect in Macosx Sierra and I had installed correctly the libfreenect(I think) but  when I want to use the Kinect I can't (camera). I want to use the wrapped of python and I always get that. error 

[image]

uni4th, 2018-10-23:
Hello, I run into the same problem. Have you solved it yet?

yavier123, 2018-10-23:
Long time ago, but yes, are you running in a mac osx or window?

uni4th, 2018-10-24:
> Long time ago, but yes, are you running in a mac osx or window? El mar., 23 de oct. de 2018 04:07, uni4th <notifications@github.com> escribió:
> […](#)

Thank you for your reply, but actually I have already solved it :)

Windows10010, 2019-03-10:
@uni4th How did you solve the problem? I have the same issue now.

leonardokidd, 2019-04-07:
This answer may be late, but you can find the solution here: https://openkinect.org/wiki/FAQ#I_get_the_following_error_accessing_microphones:_upload_firmware:_failed_to_find_firmware_file.

sumdii, 2019-07-18:
From [openkinect wiki](https://openkinect.org/wiki/FAQ#I_get_the_following_error_accessing_microphones:_upload_firmware:_failed_to_find_firmware_file. ): Kinect audio system requires a firmware to be sent at runtime. It should be found in your installation (in share/libfreenect/audios.bin) but it might be absent. Run share/libfreenect/fwfetcher.py to download the latest audio firmware file and move it to your installation.
And my installation is `/usr/local/share/libfreenect`

benjaminjacobreji, 2021-07-08:
> From [openkinect wiki](https://openkinect.org/wiki/FAQ#I_get_the_following_error_accessing_microphones:_upload_firmware:_failed_to_find_firmware_file.): Kinect audio system requires a firmware to be [...]

I used mlocate to find the fwfetcher.py file. 
I'm on Raspbian so 
```bash
$ sudo apt install mlocate
...
...
$ sudo updatedb
$ locate fwfetcher.py
/home/pi/libfreenect/src/fwfetcher.py
/usr/local/share/fwfetcher.py
$ python3 python3 /usr/local/share/fwfetcher.py
$ freenect-glview
```
And the Kinect model 1473 works in RPi3.

I also had to follow this [comment](https://github.com/OpenKinect/libfreenect/issues/543#issuecomment-492024478) to get it working.