OpenKinect/libfreenect #537: Fetch and Build instructions do not work on OSX

balandinodidonato, 2017-12-05:
### Computer info
- OS: macOS Sierra v 10.12.6
- Laptrop MacBook Pro 2017

### To Reproduce
- Lauch terminal
- follow instruction in `readme.md` file

### Expected
- The instructions fetch and build

### Actual
- Seel terminal log below
`Balandinos-MacBook-Pro:~ balandinodidonato$ git clone https://github.com/OpenKinect/libfreenect`
`Cloning into 'libfreenect'...`
`remote: Counting objects: 4792, done.`
`remote: Total 4792 (delta 0), reused 0 (delta 0), pack-reused 4792`
`Receiving objects: 100% (4792/4792), 2.10 MiB | 570.00 KiB/s, done.`
`Resolving deltas: 100% (2814/2814), done.`
`Balandinos-MacBook-Pro:~ balandinodidonato$ cd libfreenect`
`Balandinos-MacBook-Pro:libfreenect balandinodidonato$ mkdir build`
`Balandinos-MacBook-Pro:libfreenect balandinodidonato$ cd build`
`Balandinos-MacBook-Pro:build balandinodidonato$ cmake -L .. # -L lists all the project options`
`-bash: cmake: command not found`

piedar, 2017-12-05:
`brew install cmake` or https://cmake.org/download

balandinodidonato, 2017-12-05:
you are totally right there was an errro in my cmake installation

I now get this error after the `make` command

`gg1107i1-lm0005:build balandinodidonato$ make`
`[ 16%] Built target freenectstatic`
`[ 18%] Linking C shared library ../lib/libfreenect.dylib`
`Undefined symbols for architecture x86_64:`
`  "_libusb_get_parent", referenced from:`
`      _fnusb_find_sibling_device in usb_libusb10.c.o`
`ld: symbol(s) not found for architecture x86_64`
`clang: error: linker command failed with exit code 1 (use -v to see invocation)`
`make[2]: *** [lib/libfreenect.0.6.0.dylib] Error 1`
`make[1]: *** [src/CMakeFiles/freenect.dir/all] Error 2`
`make: *** [all] Error 2`
`gg1107i1-lm0005:build balandinodidonato$ `

piedar, 2017-12-05:
It's probably linking against system libusb, which is too old.  That's why we recommend

    brew install libfreenect

which links all the dependencies automatically.

If you really want to continue manually, you'll need to compile or otherwise install libusb-1.0.18 or newer and then configure libfreenect to find it.

    cmake -L .. -DLIBUSB_1_LIBRARY:FILEPATH=/path/to/custom/libusb-1.0.so