OpenKinect/libfreenect #543: freenect_glview does not work with kinect 1473

josuerocha, 2017-12-28:
I installed freenect in Ubuntu 16.04 with **audio support** using the following script:

git clone https://github.com/OpenKinect/libfreenect
cd libfreenect
mkdir build
cd build
cmake .. -L -DBUILD_AUDIO=ON
make
make install

However, it ouputs the following error:

git clone https://github.com/OpenKinect/libfreenect
cd libfreenect
mkdir build
cd build
cmake .. -L -DBUILD_AUDIO=ON
make
make install

I am using Kinect 1473. Please help.

piedar, 2017-12-28:
Please try `freenect-camtest`, which [uses only the camera](https://github.com/OpenKinect/libfreenect/blob/master/examples/camtest.c#L73).  

The root cause is probably missing audio firmware.  Model 1473 uses the audio device to control the motor, and needs firmware for the audio.  I am working on a firmware fix, but I still can't get it to work with 1473.  In the meantime, I'm thinking of printing out warnings when audio/motor fails to load instead of stopping the program.

smokhov, 2017-12-29:
FWIW, recently using `fwfetcher.py` to get `audios.bin` and using it with my 1473 on EL6 (CentOS) and OS X worked for us for all example programs.

smokhov, 2017-12-29:
(That was in #533)

piedar, 2017-12-30:
Hmm, removing the call libusb_reset_device() makes it work for me.

```
diff --git a/src/usb_libusb10.c b/src/usb_libusb10.c
index cfa6166..cc3da6a 100644
--- a/src/usb_libusb10.c
+++ b/src/usb_libusb10.c
@@ -354,7 +354,7 @@ FN_INTERNAL int fnusb_keep_alive_led(freenect_context* ctx, libusb_device* audio
 
  // we need to do this as it is possible that the device was not closed properly in a previous session
  // if we don't do this and the device wasn't closed properly - it can cause infinite hangs on LED and TILT functions
- libusb_reset_device(audioHandle);
+ //libusb_reset_device(audioHandle);
  libusb_close(audioHandle);
 
  res = libusb_open(audio, &audioHandle);
```

Jonah-Chng, 2018-01-04:
I applied the fix by setting:
"libusb_reset_device(audioHandle); "  to "//libusb_reset_device(audioHandle);"

but there is still an error, when using sudo freenect-camtest:

Failed to submit isochronous transfer 0: -1
Failed to submit isochronous transfer 1: -1
Failed to submit isochronous transfer 2: -1
Failed to submit isochronous transfer 3: -1
Failed to submit isochronous transfer 4: -1
Failed to submit isochronous transfer 5: -1
Failed to submit isochronous transfer 6: -1
Failed to submit isochronous transfer 7: -1
Failed to submit isochronous transfer 8: -1
Failed to submit isochronous transfer 9: -1
Failed to submit isochronous transfer 10: -1
Failed to submit isochronous transfer 11: -1
Failed to submit isochronous transfer 12: -1
Failed to submit isochronous transfer 13: -1
Failed to submit isochronous transfer 14: -1
Failed to submit isochronous transfer 15: -1
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
Failed to submit isochronous transfer 0: -1
Failed to submit isochronous transfer 1: -1
Failed to submit isochronous transfer 2: -1
Failed to submit isochronous transfer 3: -1
Failed to submit isochronous transfer 4: -1
Failed to submit isochronous transfer 5: -1
Failed to submit isochronous transfer 6: -1
Failed to submit isochronous transfer 7: -1
Failed to submit isochronous transfer 8: -1
Failed to submit isochronous transfer 9: -1
Failed to submit isochronous transfer 10: -1
Failed to submit isochronous transfer 11: -1
Failed to submit isochronous transfer 12: -1
Failed to submit isochronous transfer 13: -1
Failed to submit isochronous transfer 14: -1
Failed to submit isochronous transfer 15: -1
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00

and when using sudo freenect-glview:

Kinect camera test
Number of devices found: 1
Found sibling device [same parent]
Failed to set the LED of K4W or 1473 device: -1
Found sibling device [same parent]
Could not open device: -1
Could not open device

Any idea how I can fix this?

wpumacay, 2018-02-03:
Hi
I'm running into the same trouble. 

Failed to submit isochronous transfer 0: -1
Failed to submit isochronous transfer 1: -1
Failed to submit isochronous transfer 2: -1
Failed to submit isochronous transfer 3: -1
Failed to submit isochronous transfer 4: -1
Failed to submit isochronous transfer 5: -1
Failed to submit isochronous transfer 6: -1
Failed to submit isochronous transfer 7: -1
Failed to submit isochronous transfer 8: -1
Failed to submit isochronous transfer 9: -1
Failed to submit isochronous transfer 10: -1
Failed to submit isochronous transfer 11: -1
Failed to submit isochronous transfer 12: -1
Failed to submit isochronous transfer 13: -1
Failed to submit isochronous transfer 14: -1
Failed to submit isochronous transfer 15: -1
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00

I'm testing with os = archlinux, built freenect from the github repository and I'm using libsub1.0.21-2 ( from arch repository ). The other samples that don't use the camera kind of work ( after applying the fix suggested lines above about removing libusb_reset_device ). The freenect-micview sample works; i can see the sound waves in the screen. The freenect tilt demo also works. Also tested it using usb 2.0 and 3.0.
The samples that use the camera show that error and show me the window where the frames should be, but no image is shown.

I tried a kinectv2 using libfreenect2 and it worked normally, just my kinectv1 with libfreenect doesn't work.

Is there anything you could suggest me to fix this issue ?

Thanks.

Jonah-Chng, 2018-02-06:
Hi all and @wpumacay,

I fixed the problem however I reinstalled ROS for it to work haha!

But, I am currently using libfreenect_launch and it works just fine for me!

@wpumacay what are you tryign to do with kinect?

wpumacay, 2018-02-06:
Hi.

I'm trying to build the lib from source and make some other cameras we have in the lab work in linux. So far, the kinectv1 we have is the only one giving us trouble both in ubuntu and arch, and I wouldn't like to switch to windows just to test it there.

So, you had to just reinstall ROS for it to work?. In my case I'm not using ROS but libfreenect from the repo. By a quick search, it appears that the ros packages use a wrapper ROS-freenect-stack and use version d1d19df8957f8d63a562b310413da15d3e6ef7a1 of libfreenect from way back 2012. I will give that version a try and see if it works. Thanks for the hint 👍

skribtsov, 2018-02-21:
after installing the audios.bin received by fwfetcher.py I was able to get rid of the message: "Failed to set the LED of K4W or 1473 device: -1", however "Failed to submit isochronous transfer 15: -1" are still there. BTW it was not clear how to install the audios.bin. I did it by running an example program that works with microphones 

freenect-micview

So, how to make the depth camera work?!

black1night, 2018-06-04:
I installed freenect through ROS and could not display camera information and depth information
kinect model 1473

gatali, 2018-08-12:
i have got same problem my model is 1473 and my freenect-camtest work right. But my problem at below. Please help me how can i solve this?

Kinect camera test
Number of devices found: 1
Found sibling device [same parent]
Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_NO_DEVICE
Found sibling device [same parent]
Could not open device: LIBUSB_ERROR_NO_DEVICE
Could not open device

ronthalanki, 2018-11-19:
i have the same issue as gatali, any updates on this?

gatali, 2018-11-23:
I solved the problem and shared my YouTube channel  

https://www.youtube.com/watch?v=eOUjvwyRz9U&t=3s

StoneLabs, 2019-05-13:
I know this is an old thread but just in case anyone is still facing the issue:
Commenting out the following [line:](https://github.com/OpenKinect/libfreenect/blob/aca6046a5bc1bc5bbe7f2ae8c0085464e7ebd9f1/src/usb_libusb10.c#L443) fixed all issues we had with our 1473.

frederico-klein, 2020-03-19:
@StoneLabs solution worked for me (after downloading the audio firmware `python ./src/fwfetcher.py` i didn't know I had to do that)

yasiupl, 2020-06-12:
What did it for me, was running `freenect-micview` and `freenect-camtest` before `freenect-glview`.