OpenKinect/libfreenect #547: Kinect 1517 - Tilt, Motor, Leds and Audio work. But no frames coming from freenect-camtest

pedromundo, 2018-01-20:
As some of the demos directly mention the K4W I recently acquired a kinect 1517 (despite all the previous comments about it being a nasty piece of work) and have been having trouble getting depth information from it. I have an application which relies on the kinect to capture depth images and have tried most of the things listed in past issues (even some of the things that I've now realized to be integrated back on the master branch ha) and nothing seemed to work.

Running freenect-micview seems to do something. Without prior micview execution, running my application results in:

`Could not find device sibling
ERROR: could not open device`

After running micview, which claims to detect and upload a new firmware, the application starts ok but hangs with empty windows as no frames are ever received, I've confirmed that no frames arrive via camtest. This behavior is similar to what I experienced with my old kinect before that fix regarding new usbcore/isochronous transfers was rolled out.

Anyone have any ideas?

piedar, 2018-01-23:
Sadly I don't have a model 1517 anymore...  Anyway if the USB transfers are hanging maybe this will give more info:

    LIBUSB_DEBUG=3 freenect-camtest

pedromundo, 2018-01-24:
the only additional info I get with LIBUSB_DEBUG is a bunch of:

`libusb: error [submit_iso_transfer] submiturb failed error -1 errno=2`

Both before and after 'flashing' the new firmware with 'micview'. I have also noticed that pressing 'n' on glview says that 'Near mode is only supported by K4W', as if my unit wasn't a K4W one. Do you guys have any other ideas to help troubleshoot this?

Could this be due to a defective unit or something else on the sellers end (i.e. plugging it into an xbox to 'test')?

closed 2022-08-20 by pedromundo