OpenKinect/libfreenect #548: more straightforward way of getting an active device's serial number?

svenevs, 2018-02-11:
I'm using the C++ wrappers, so I have some class like

```cpp
class Kinect360_Impl : public Freenect::FreenectDevice {
public:
    Kinect360_Impl(freenect_context *_ctx, int _index)
        : Freenect::FreenectDevice(_ctx, _index)
        , mCtx(_ctx)
        , mIndex(_index) {
        setDepthFormat(FREENECT_DEPTH_MM, FREENECT_RESOLUTION_MEDIUM);
    }
```

and can then emulate the loop here

https://github.com/OpenKinect/libfreenect/blob/83e57e1318cc64c9aabac481b9e330acc1914a23/src/core.c#L194-L199

since I've stored both the index and the context, and can get the `freenect_device *` from `Kinect360_Impl->getDevice()`.  So instead of comparing strings, I'm actually just checking what index I'm at.

I looked around the API for a while, but couldn't figure out how else to do it.  Is this the only way, or is there something I'm overlooking?

If it matters, the reason I want this is for calibration purposes.  I am storing different calibrations in a single XML file so that at run-time I can check if the serial number of the active device has been calibrated (by checking this file).  If so, I default to using the calibrated intrinsics rather than the defaults.

piedar, 2018-03-06:
I looked into this briefly, and it's not easy.  It would be easy if we could always trust the camera serial, but we have to [fall back on the audio serial](https://github.com/OpenKinect/libfreenect/blob/master/src/usb_libusb10.c#L207) for K4W devices.

svenevs, 2018-03-18:
I see.  Out of curiosity, why is the fallback a problem?  Is it that the audio serial is a different length or could possibly be a "duplicate" to a real Kinect serial number?

piedar, 2018-03-19:
The audio serial itself works fine, but it's tough to get it to handle all edge cases.  Like what if the caller asked to open MOTOR only?  Unlikely, but it makes it more difficult to figure out which `libusb_device`s go with which `freenect_device`.

It appears we'd have to do another libusb query for the list of all devices and search through them to find the matching audio device, or try to do everything up-front when opening subdevices (which is rather brittle code already).  It's also necessary to call libusb_open_device() before querying the serial, which opens another door for failure.  My first attempt was complete spaghetti code.

But now that I think about it, it might work to add a `libusb_device* usb_dev` member to [`fnusb_dev`](https://github.com/OpenKinect/libfreenect/blob/master/src/usb_libusb10.h#L57).  Then [when opening subdevices](https://github.com/OpenKinect/libfreenect/blob/master/src/usb_libusb10.c#L381) we could always populate those entries, even if we're not actually going to open the associated device handles.  (We'd also have to call libusb_ref_device() to make sure the pointers don't get freed when the list gets unrefed.)  That at least solves the problem of which libusb devices go with the freenect device.

I don't have a lot of time to work on this, but I'll give it another go when I get a chance.

piedar, 2018-03-23:
Good news!  Check out #554.

closed 2018-03-23 by piedar