OpenKinect/libfreenect #550: freenect no module found

manikanthgoud559, 2018-02-11:
(cv) manikanth@jarvis:~/libfreenect/wrappers/python$ python kinect_test.py 
Traceback (most recent call last):
  File "kinect_test.py", line 2, in <module>
    import freenect
ImportError: No module named 'freenect'

jerwelborn, 2018-02-14:
Check out the bottom of the README within wrappers/python:

```
- If you get "/usr/bin/ld: cannot find -lfreenect", then it can't find the library.  It looks in '/usr/local/lib' '/usr/local/lib64' along with anything set in LD_LIBRARY_PATH.  You probably need to do sudo make install in the build dir.  If it puts it in a directory that isn't one of those 2 let me know and I'll add it as another place to look.
```

manikanthgoud559, 2018-02-15:
Actually in my machine opencv was installed in virtual terminal and
freenect module in normal terminal.
when i opened python interpreter in workon cv and import freenect it gives
no module found where as in normal terminal its importing freenect module

how can i run both in one place or share any link as i am new to linux

Thank you  so muchh

manikanthgoud559, 2018-03-17:
i have found in /usr/local/lib and set it to LD_LIBRARY_PATH and did this sudo make install in build dir in libfreenect but still it says no "freenect" module found

Klinggaard, 2018-04-06:
I had the same issue. But all I did to fix it was to do what this guy did.  [guide](https://naman5.wordpress.com/2014/06/24/experimenting-with-kinect-using-opencv-python-and-open-kinect-libfreenect/)

After that, everything worked fine. But you have to do everything

jackietom, 2018-07-10:
I didn't install freenect from source. You can use sudo apt-get install freenect on ubuntu/debian. To use python package you can install python-freenect using apt.

manikanthgoud559, 2018-07-10:
I had done same what was in guide too but the problem is open CV was
installed in envs and freenect in normal terminal even I reinstalled it but
same happens

Thanks for responding

remmel, 2021-11-14:
> I had the same issue. But all I did to fix it was to do what this guy did. [guide](https://naman5.wordpress.com/2014/06/24/experimenting-with-kinect-using-opencv-python-and-open-kinect-libfreenect/)
> 
> After that, everything worked fine. But you have to do everything

Exactly, especially: 
```
cd libfreenect/wrappers/python
sudo python setup.py install
``

DunZek, 2021-11-21:
Doing `sudo python setup.py install` in `libfreenect/wrappers/python` gives

```
running install
running build
running build_ext
skipping 'freenect.c' Cython extension (up-to-date)
building 'freenect' extension
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.9 -c freenect.c -o build/temp.linux-x86_64-3.9/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /usr/local/lib/python3.9/dist-packages/numpy/core/include
In file included from /usr/local/lib/python3.9/dist-packages/numpy/core/include/numpy/ndarraytypes.h:1969,
                 from /usr/local/lib/python3.9/dist-packages/numpy/core/include/numpy/ndarrayobject.h:12,
                 from /usr/local/lib/python3.9/dist-packages/numpy/core/include/numpy/arrayobject.h:4,
                 from freenect.c:353:
/usr/local/lib/python3.9/dist-packages/numpy/core/include/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   17 | #warning "Using deprecated NumPy API, disable it with " \
      |  ^~~~~~~
freenect.c: In function ‘PyInit_freenect’:
freenect.c:10210:30: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
10210 |   __pyx_type_8freenect_CtxPtr.tp_print = 0;
      |                              ^
freenect.c:10214:30: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
10214 |   __pyx_type_8freenect_DevPtr.tp_print = 0;
      |                              ^
freenect.c:10218:32: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
10218 |   __pyx_type_8freenect_StatePtr.tp_print = 0;
      |                                ^
freenect.c: In function ‘__Pyx_ParseOptionalKeywords’:
freenect.c:11450:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:106,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘PyUnicode_AsUnicode’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:106,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:580:45: note: declared here
  580 | Py_DEPRECATED(3.3) PyAPI_FUNC(Py_UNICODE *) PyUnicode_AsUnicode(
      |                                             ^~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:106,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:106,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘PyUnicode_AsUnicode’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:106,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:580:45: note: declared here
[... 43 lines cut ...]
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
freenect.c:11466:25: warning: ‘PyUnicode_AsUnicode’ is deprecated [-Wdeprecated-declarations]
11466 |                         (PyUnicode_GET_SIZE(**argname) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                         ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
[... cut ...]
11466 |                         (PyUnicode_GET_SIZE(**argname) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                         ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:106,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1
```
Why is this?

mhered, 2021-12-16:
Found the same issue after following the instructions described [here](https://naman5.wordpress.com/2014/06/24/experimenting-with-kinect-using-opencv-python-and-open-kinect-libfreenect/)
However I noticed a message while installing `cython` that the module `python-is-python3` would be removed.
So I modified the following line:
```bash
sudo python3 setup.py install
```
et voilà it worked!
Hope it helps

armorvx, 2022-02-17:
nice try @mhered ... version definition of python || python3 isn´t working for compiling... 

```
/usr/bin/python3 setup.py install

running install
/usr/local/lib/python3.9/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/local/lib/python3.9/dist-packages/setuptools/command/easy_install.py:160: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running bdist_egg
running egg_info
writing freenect.egg-info/PKG-INFO
writing dependency_links to freenect.egg-info/dependency_links.txt
writing top-level names to freenect.egg-info/top_level.txt
reading manifest file 'freenect.egg-info/SOURCES.txt'
writing manifest file 'freenect.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_ext
skipping 'freenect.c' Cython extension (up-to-date)
building 'freenect' extension
arm-linux-gnueabihf-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/usr/include/python3.9 -c freenect.c -o build/temp.linux-armv7l-3.9/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /usr/local/lib/python3.9/dist-packages/numpy/core/include
In file included from /usr/include/python3.9/numpy/ndarraytypes.h:1822,
                 from /usr/include/python3.9/numpy/ndarrayobject.h:12,
                 from /usr/include/python3.9/numpy/arrayobject.h:4,
                 from freenect.c:353:
/usr/include/python3.9/numpy/npy_1_7_deprecated_api.h:17:2: warning: #warning "Using deprecated NumPy API, disable it with " "#define NPY_NO_DEPRECATED_API NPY_1_7_API_VERSION" [-Wcpp]
   17 | #warning "Using deprecated NumPy API, disable it with " \
      |  ^~~~~~~
freenect.c: In function ‘PyInit_freenect’:
freenect.c:10210:30: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
10210 |   __pyx_type_8freenect_CtxPtr.tp_print = 0;
      |                              ^
freenect.c:10214:30: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
10214 |   __pyx_type_8freenect_DevPtr.tp_print = 0;
      |                              ^
freenect.c:10218:32: error: ‘PyTypeObject’ {aka ‘struct _typeobject’} has no member named ‘tp_print’
10218 |   __pyx_type_8freenect_StatePtr.tp_print = 0;
      |                                ^
freenect.c: In function ‘__Pyx_ParseOptionalKeywords’:
freenect.c:11450:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:97,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘PyUnicode_AsUnicode’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:97,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:580:45: note: declared here
  580 | Py_DEPRECATED(3.3) PyAPI_FUNC(Py_UNICODE *) PyUnicode_AsUnicode(
      |                                             ^~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
      |                     ^
In file included from /usr/include/python3.9/unicodeobject.h:1026,
                 from /usr/include/python3.9/Python.h:97,
                 from freenect.c:16:
/usr/include/python3.9/cpython/unicodeobject.h:446:26: note: declared here
  446 | static inline Py_ssize_t _PyUnicode_get_wstr_length(PyObject *op) {
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~
freenect.c:11450:21: warning: ‘_PyUnicode_get_wstr_length’ is deprecated [-Wdeprecated-declarations]
11450 |                     (PyUnicode_GET_SIZE(**name) != PyUnicode_GET_SIZE(key)) ? 1 :
[... 99 lines cut ...]
      |                         ^~~~~~~~~
      |                         curexc_value
freenect.c:11604:22: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
11604 |     tmp_tb = tstate->exc_traceback;
      |                      ^~~~~~~~~~~~~
      |                      curexc_traceback
freenect.c:11605:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_type’; did you mean ‘curexc_type’?
11605 |     tstate->exc_type = type;
      |             ^~~~~~~~
      |             curexc_type
freenect.c:11606:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_value’; did you mean ‘curexc_value’?
11606 |     tstate->exc_value = value;
      |             ^~~~~~~~~
[... cut ...]
freenect.c:11607:13: error: ‘PyThreadState’ {aka ‘struct _ts’} has no member named ‘exc_traceback’; did you mean ‘curexc_traceback’?
11607 |     tstate->exc_traceback = tb;
      |             ^~~~~~~~~~~~~
      |             curexc_traceback
error: command '/usr/bin/arm-linux-gnueabihf-gcc' failed with exit code 1
```

I need to use the python freenect library on following system...
Linux version 5.10.92-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0

anybody an idea which changes fix that?

cuciureansergiu, 2023-02-09:
> 

Use python 3.8

TolyaDenisenko, 2024-07-20:
Hello people!! 
Help me please ))

sudo python setup.py install
/home/anatoly/libfreenect/wrappers/python/setup.py:2: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.core import setup
running install
running build
running build_ext
building 'freenect' extension
creating build
creating build/temp.linux-x86_64-3.10
x86_64-linux-gnu-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.10 -c freenect.c -o build/temp.linux-x86_64-3.10/freenect.o -fPIC -I ../../include/ -I /usr/include/libusb-1.0/ -I /usr/local/include/libusb-1.0 -I /usr/local/include -I ../c_sync/ -I /usr/lib/python3/dist-packages/numpy/core/include
cc1: fatal error: freenect.c: No such file or directory
compilation terminated.
error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1