OpenKinect/libfreenect #553: Python wrapper fails to built against opencv 3.4

AdamWill, 2018-03-05:
Build of libfreenect against opencv 3.4.1 [failed for Fedora](https://koji.fedoraproject.org/koji/taskinfo?taskID=25506808). This was the error:

    cd /builddir/build/BUILD/libfreenect-ddf092cce09ab4bc97529827a061f1a54a046a17/build/wrappers/python && /usr/bin/cython -3 -o freenect3.c /builddir/build/BUILD/libfreenect-ddf092cce09ab4bc97529827a061f1a54a046a17/wrappers/python/freenect.pyx
    make -f OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/build.make OpenNI2-FreenectDriver/CMakeFiles/FreenectDriver.dir/depend
    In file included from /usr/include/opencv2/core/types_c.h:59,
                     from /usr/include/opencv2/core/core_c.h:48,
                     from /usr/include/opencv/cv.h:63,
                     from /builddir/build/BUILD/libfreenect-ddf092cce09ab4bc97529827a061f1a54a046a17/wrappers/opencv/libfreenect_cv.h:7,
                     from /builddir/build/BUILD/libfreenect-ddf092cce09ab4bc97529827a061f1a54a046a17/wrappers/opencv/libfreenect_cv.c:3:
    /usr/include/opencv2/core/cvdef.h:485:1: error: unknown type name 'namespace'
     namespace cv {
     ^~~~~~~~~
    /usr/include/opencv2/core/cvdef.h:485:14: error: expected '=', ',', ';', 'asm' or '__attribute__' before '{' token
     namespace cv {
                  ^

Digging into this I found an [upstream issue](https://github.com/opencv/opencv/issues/10963) with some discussion of this, where [upstream states](https://github.com/opencv/opencv/issues/10963#issuecomment-369259044) "OpenCV 3.x doesn't not support C compilation mode officially. There are still legacy C API calls (from OpenCV 1.x) but they should be compiled in C++ compilation mode."

AdamWill, 2018-03-05:
Also note this from upstream: "Even if you succeeded to build code with C compiler (and avoid cvRound problem: #6585 <#6585>) it would lead to terrible behavior like #6221 <#6221> or silent crashes in case of raised C++ exceptions."

So it really sounds like the Python wrapper should use C++ if it wants to work with opencv 3.x...

piedar, 2018-03-06:
Are you sure it's the python wrapper?  It looks to me like the [opencv wrapper](https://github.com/OpenKinect/libfreenect/tree/master/wrappers/opencv) is what's failing, but it's hard to tell because `make -j6` makes the build output out of order.

AdamWill, 2018-03-08:
Well, I was assuming it was when building the bits of the Python wrapper that *use* the opencv wrapper. But if the opencv wrapper itself builds as pure C it likely would hit this too. So it could just be that, yeah. I can re-run the compile with parallelization disabled to get clearer output if you like, but not sure if it's necessary, the basic issue is clear enough (you aren't supposed to compile as pure C against opencv 3).

piedar, 2018-03-23:
Looks like you worked around it by setting `-DBUILD_CV=OFF`.  The CV wrapper ought to be deprecated anyway, and replaced with a small example showing how to convert freenect frames to `cv::Mat`.

armandok, 2018-05-20:
I'm having the same issue on Ubuntu 18.04 with OpenCV 3.4.1.

PR #555 (2018-05-26): Opencv cpp api
This PR includes fixes to use the OpenCV api in C++ mode, and fixes to some error handling issues I found when trying to run "freenect-cvdemo" without a kinect plugged into the PC (the application would segfault instead of notifying the user that it could not open a camera.)  It should fix #553.
closed 2018-05-26 by piedar