OpenKinect/libfreenect #556: Skeleton tracking

josuerocha, 2018-05-28:
Is there any algorithm available for skeletal tracking? I mean, like the one in KinectSDK, that uses random decision trees.

piedar, 2018-05-28:
It's a difficult problem, and most solutions are proprietary.  [Microsoft's random decision tree](https://www.microsoft.com/en-us/research/publication/real-time-human-pose-recognition-in-parts-from-a-single-depth-image/) method is fairly well documented, but requires mo-cap data and [expensive animation software](https://www.autodesk.com/products/motionbuilder/overview) to generate the training data.

There are some open-source contenders, but none appear to use the state-of-the-art approach.

- https://github.com/fpeder/XKin
- https://github.com/joaquimrocha/Skeltrack

josuerocha, 2018-05-28:
Thank you for the information.