OpenKinect/libfreenect #557: LIBUSB_ERROR_IO audio device K4W (1517)

ac130kz, 2018-06-04:
I've built libfreenect and libusb from sources. Most of examples exit with generic (-4 or -6) error (surprisingly, regview works fine). Even the examples which are within the ROS present me the same error pattern. I don't need the audio device for my project at all, is there a way to fully disable it within libfreenect or fix my issue?

> user@workstation:~/libfreenect/build/bin$ LIBUSB_DEBUG=3 ./freenect-cvdemo
> libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/110: No such device
> Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO
> libusb: error [_get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/110: No such device
> Could not open device: LIBUSB_ERROR_IO
> Error: Invalid index [0]
> Error: Kinect not connected?

Before execution: lsusb | grep Microsoft
> Bus 003 Device 111: ID 045e:02bf Microsoft Corp. Kinect for Windows NUI Camera
> Bus 003 Device 113: ID 045e:02be Microsoft Corp. Kinect for Windows NUI Audio
> Bus 003 Device 108: ID 045e:02c2 Microsoft Corp. Kinect for Windows NUI Motor

Just after exit: lsusb | grep Microsoft
> Bus 003 Device 111: ID 045e:02bf Microsoft Corp. Kinect for Windows NUI Camera
> Bus 003 Device 108: ID 045e:02c2 Microsoft Corp. Kinect for Windows NUI Motor

10 secs later: lsusb | grep Microsoft
> Bus 003 Device 111: ID 045e:02bf Microsoft Corp. Kinect for Windows NUI Camera
> Bus 003 Device 116: ID 045e:02be Microsoft Corp. Kinect for Windows NUI Audio
> Bus 003 Device 108: ID 045e:02c2 Microsoft Corp. Kinect for Windows NUI Motor

dewaldcrous, 2018-06-22:
I have the exact same problem, have you found a solution to the problem at hand by any chance ?

ac130kz, 2018-06-22:
Unfortunately, no. Seems like a K4W firmware/compatibility related issue, as I saw people on this issue tracker suggesting to add lines in order to upload audios.bin on the device, which did not work for me (maybe it will work for you, who knows). What is even more interesting is that I'm currently writing own multi-camera calibration tool, because ccalib and MATLAB options offer horribly wrong results...

Lankardy, 2018-06-22:
Enviado pelo meu Windows Phone

Enviada em: ‎22/‎06/‎2018 12:15
Para: OpenKinect/libfreenect<mailto:libfreenect@noreply.github.com>
Cc: Subscribed<mailto:subscribed@noreply.github.com>

thatmadhacker, 2018-12-06:
I am having this same issue, I am running the examples/hiview.c program and it gives me an error that looks like this 

`Kinect camera test`

`Number of devices found: 1`

`Found sibling device [same parent]`

`Failed to set the LED of K4W or 1473 device: -1`

`Found sibling device [same parent]`

`Could not open device: -1`

`Could not open device`

and when I run `lsusb` while the program is executing I get the same behavior

linmeeka, 2018-12-17:
> I am having this same issue, I am running the examples/hiview.c program and it gives me an error that looks like this
>
> `Kinect camera test`
>
> `Number of devices found: 1`
>
> `Found sibling device [same [...]

Hi,i got the same issue,and when i run ` freenect-micview`,met the error:

```
Number of devices found: 1
Trying to open ./audios.bin as firmware...
Trying to open /home/qizhang/.libfreenect/audios.bin as firmware...
Trying to open /usr/local/share/libfreenect/audios.bin as firmware...
Trying to open /usr/share/libfreenect/audios.bin as firmware...
Trying to open ./../Resources/audios.bin as firmware...
upload_firmware: failed to find firmware file.
upload_firmware failed: -2
Could not open device
```
seems like there are some problems with audio firmware.
I don't know how to fix it ,do you have any idea?

thatmadhacker, 2018-12-17:
Did you download the firmware file?

thatmadhacker, 2018-12-17:
Run src/fwfetcher.py and then run mictest

linmeeka, 2018-12-17:
> Run src/fwfetcher.py and then run mictest

Thank you!
I run fwfetcher.py and downloaded audios.bin.
But when i run ```freenect-glview```, i got these error:
```
./freenect-glview 
Kinect camera test
Number of devices found: 1
Found sibling device [same parent]
Found sibling device [same parent]
GL thread
[Stream 70] Negotiated packet size 1920
Failed to submit isochronous transfer 0: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 1: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 2: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 3: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 4: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 5: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 6: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 7: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 8: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 9: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 10: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 11: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 12: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 13: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 14: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 15: LIBUSB_ERROR_IO
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00

```
Could you kindly help me?

thatmadhacker, 2018-12-17:
That doesn't look like a fatal error so it should not matter

terminator-ger, 2019-01-11:
I have the exact same Problem, using a Kinect 1517 on a Jetson TX2 (Ubuntu 16.04). After Uploading the Firmware via freenect-micview, entering freenect-glview gives me:

> Found sibling device [same parent]
Found sibling device [same parent]
[Stream 70] Negotiated packet size 1920
Failed to submit isochronous transfer 0: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 1: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 2: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 3: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 4: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 5: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 6: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 7: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 8: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 9: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 10: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 11: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 12: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 13: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 14: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 15: LIBUSB_ERROR_IO
write_register: 0x0105 <= 0x00
write_register: 0x0006 <= 0x00
write_register: 0x0012 <= 0x03
write_register: 0x0013 <= 0x01
write_register: 0x0014 <= 0x1e
write_register: 0x0006 <= 0x02
write_register: 0x0017 <= 0x00
[Stream 80] Negotiated packet size 1920
Failed to submit isochronous transfer 0: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 1: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 2: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 3: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 4: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 5: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 6: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 7: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 8: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 9: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 10: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 11: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 12: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 13: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 14: LIBUSB_ERROR_IO
Failed to submit isochronous transfer 15: LIBUSB_ERROR_IO
write_register: 0x000c <= 0x00
write_register: 0x000d <= 0x01
write_register: 0x000e <= 0x1e
write_register: 0x0005 <= 0x01
write_register: 0x0047 <= 0x00
Kinect camera test
Number of devices found: 1
GL thread
'w' - tilt up, 's' - level, 'x' - tilt down, '0'-'6' - select LED mode, '+' & '-' - change IR intensity 
'f' - change video format, 'm' - mirror video, 'o' - rotate video with accelerometer 
'e' - auto exposure, 'b' - white balance, 'r' - raw color, 'n' - near mode (K4W only) 

But the GL-Window from glview stays plain black, and shows neither depth nor rgb.

jonasborn, 2019-02-27:
Also facing this problem when building from source:

> -- Operating system is Linux
> -- Got System Processor x86_64
> -- Linux x86_64 Detected
> -- libfreenect will be installed to /usr/local
> -- Headers will be installed to /usr/local/include/libfreenect
> -- Libraries will be installed to /usr/local/lib
> -- Found libusb-1.0:
> --  - Includes: /usr/include/libusb-1.0
> --  - Libraries: /home/jonas/test/libusb-1.0.22/libusb/.libs/libusb-1.0.so
> -- Configuring done
> -- Generating done
> -- Build files have been written to: /home/jonas/test/libfreenect/build
> [  1%] Linking C shared library ../lib/libfreenect.so
> [ 16%] Built target freenect
> [ 33%] Built target freenectstatic
> [ 35%] Linking C shared library ../../lib/libfreenect_sync.so
> ...

Building works, but the following message appears running camtest:

> Found sibling device [same parent]
> Failed to set the LED of K4W or 1473 device: LIBUSB_ERROR_IO
> [Stream 70] Negotiated packet size 1920
> Failed to submit isochronous transfer 0: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 1: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 2: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 3: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 4: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 5: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 6: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 7: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 8: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 9: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 10: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 11: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 12: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 13: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 14: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 15: LIBUSB_ERROR_IO
> write_register: 0x0105 <= 0x00
> write_register: 0x0006 <= 0x00
> write_register: 0x0012 <= 0x03
> write_register: 0x0013 <= 0x01
> write_register: 0x0014 <= 0x1e
> write_register: 0x0006 <= 0x02
> write_register: 0x0017 <= 0x00
> [Stream 80] Negotiated packet size 1920
> Failed to submit isochronous transfer 0: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 1: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 2: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 3: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 4: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 5: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 6: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 7: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 8: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 9: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 10: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 11: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 12: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 13: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 14: LIBUSB_ERROR_IO
> Failed to submit isochronous transfer 15: LIBUSB_ERROR_IO
> write_register: 0x000c <= 0x00
> write_register: 0x000d <= 0x01
> write_register: 0x000e <= 0x1e
> write_register: 0x0005 <= 0x01
> write_register: 0x0047 <= 0x00
> ^CShutting down
> write_register: 0x0006 <= 0x00
> write_register: 0x0005 <= 0x00
> Done!

> ...
> DISTRIB_DESCRIPTION="Ubuntu 18.04.2 LTS"
> ...

Going to try a older version of libusb tomorrow
Btw. the packaged Version is working without any problems, expect the missing hd view functionality

betatester, 2019-03-12:
I've installed  all the package on a new machine with Ubuntu 18.04
and I've this problem discribed LIBUSB_ERROR_IO with 1473 kinect, that working on an other machine with Ubuntu 16.04
I've made src/fwfetcher.py : nothing new
I change with an 1414 kinect: that's work well
don't know why but "arf!"

thatmadhacker, 2019-03-12:
I got that error too, it is non fatal error and did not affect usage of my kinect when it happened