OpenKinect/libfreenect #559: Can I use libfreenect to get my kinect v1's intrinsics?

qhykwsw, 2018-06-27:
I want to do hand pose estimation with my kinect v1. But I'm not sure about the intrinsics of the depth camera, so can libfreenect output or calibrate it.

piedar, 2018-06-27:
Does [freenect_registration.h](https://github.com/OpenKinect/libfreenect/blob/master/include/libfreenect_registration.h) have what you need?  It has fields for things like "distance between IR camera and IR emitter," and "focal length of the IR camera".

qhykwsw, 2018-06-29:
I have tried your method, but it seems that the "freenect_zero_plane_info" in the "freenect_registration.h" didn't store the true camera's intrinsics. The outpust are like this:

"dcmos_emitter_distance: 7.500000
dcmos_rcmos_distance:   2.300000
reference_distance:     120.000000
reference_pixel_size:   0.104200"

As you see, the focal length of the IR camera is only 120mm, which if far lower than the normal value.

closed 2018-07-08 by qhykwsw